platformio/platformio-home #12: New feature request - Board specific examples code not available

toogooda, 2019-10-18:
I was asked to add this here, after posting in forum. Love the tool and impressed how active the project is.

_________

Hi all,

New to PlatformIO but using Arduino IDE for a while, I am really impressed whith how all my board just work, I was frustrated how Some worked on Windows others on Mac so I am sold on this new IDE :slight_smile:

Examples are the thing I am struggling to get my head around in the Arduino IDE all Platform/Board/Library examples showed based on selected borad, which made it very quick to get new stuff going.

From what I can tell PlatformIO examples work this way:

List item Platform examples show in home screen under examples.

List item Library examples are available in library manager but project has to be manually setup for it.

List item Board specific examples ???

Is the board examples available anywhere? I can’t find them i.e. I have a Heltec Web Kit 32 which is an ESP32 it works fine for ESP32 generic stuff but I can’t find the examples for its embeded OLED screan etc that is available for the board in Ardiuno IDE in fact there is not specific examples as its not a new platform just a new board.

Can anyone please help?

Thanks.

prampec, 2021-12-14:
I still miss this feature.
Unfortunately this is quite important one as we are missing a feature over Arduino.
I can imagine a task called "Show examples" (under project tasks and selecting a target environment) where a page is opened containing all the examples organized by Platform/Board/Used libraries

MattMgn, 2023-02-21:
Would be indeed very convenient, and would facilitate the transition from Arduino Ide to Platformio

bruceoberg, 2023-09-04:
i have an adafruit feather, plus some wings (daughterboards). getting at all the adafruit/arduino/esp32 examples is the only reason i go back to using the arduino ide. i'd much prefer to be able to browse them via platform io.

Robiv8, 2024-03-09:
wow! 4 jears on the TODO list. Impresiv

toogooda, 2024-03-11:
Looks like it was removed in 2019, I gave up and use new Arduino IDE for any examples, usually end up finishing the project there also these days.

umop3plsdn, 2024-11-15:
still a todo is kinda sad