platformio/platformio-home #14: Allow name of 'Import Arduino Project' to be specified (PIO Home)

pfeerick, 2019-10-18:
- [x] **Feature Request**.

------------------------------------------------------------------

In a similar vein to platformio/platformio-core#1312, it would be nice if the 'Import Arduino Project' function in PIO Home would let you specify the name of the project folder that will be created - perhaps as a field that is pre-filled with the date-time-board string that is currently used, or with the name of the Arduino sketch. 

This is to prevent the hassle of having to remove the sketch from the IDE being used (i.e. VSCode) and navigate to the path of the imported sketch in order to rename it, and then re-add it. 

Related to https://community.platformio.org/t/how-to-rename-a-project/2592/19?u=pfeerick

bjarneschulz, 2020-05-01:
I'm just started with platformio after using Arduino IDE for years. Importing my old projects was the first thing to try. I really miss the possibility to keep /reuse the name of my old project for the folder of the imported project.
I hope this feature will be implemented soon...

bentwookie, 2020-06-27:
Same. How it works now is very confusing and looks really sloppy. What project are you working on? Oh, you know good ol' 200627-153808!