platformio/platformio-home #24: Libraries: Builtin

legionaryu, 2019-10-28:
TypeError: Cannot read property 'map' of undefined
    at t.value (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:18:213524)
    at Zi (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:70772)
    at Ji (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:70567)
    at Ds (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:105897)
    at kl (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:90221)
    at Rl (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:90146)
    at El (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:87486)
    at http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:45816
    at t.unstable_runWithPriority (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:42:3465)
    at Ka (http://127.0.0.1:8010/b53d63f92df2c62861fd.min.js:34:45525)

pfeerick, 2019-10-29:
Error message looks supiciously similar to that of #23 ... possibly related?

ivankravets, 2019-11-01:
Duplicate of https://github.com/platformio/platformio-home/issues/23

closed 2019-11-01 by ivankravets

ivankravets, 2019-11-01:
This is actually a bug, I reproduced it. Thanks!

commit 1aec3a38 (2019-11-01): Fix an error when built-in library does not have keywords // Resolve #24