platformio/platformio-home #25: Inspect defect report - support inline configuration

pfeerick, 2019-10-29:
Now that `--inline-suppr` is[ enabled by default for a call to cppcheck with the `pio check` command](https://github.com/platformio/platformio-core/commit/5573c3871cc4d16d945478ae12e3f12e1dd578fa), the PIO Home inspect process needs that also, as it still shows the warnings that have been otherwise turned off. 

http://docs.platformio.org/en/latest/plus/check-tools/cppcheck.html#suppressing-warnings

Another one for @elyulka ... sorry! 😮

valeros, 2019-10-29:
Hi @pfeerick ! Could you please explain in more detail, you used `// cppcheck-suppress warningID` but warning is still reported? Do you see this warning if you run `pio check` in command line? Internally PIO Home uses the same `pio check` command with `--inline-suppr` so ignored warnings shouldn't be visible in Inspector report.

pfeerick, 2019-10-29:
When I run pio check, the warning is gone... i.e. 

[image]

And when I run PIO Inspect it's ... um... gone? I must have had the report open or something... I swear it wasn't working before! 😆 

I'll shut up now! 👣 👣

closed 2019-10-29 by pfeerick