platformio/platformio-home #45: Internal Store Exception

johanliebs, 2019-11-08:
TypeError: t.map(...).filter(...).flat is not a function
    at y (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:6:55402)
    at b (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:6:55466)
    at Function.mapToProps (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:18:178950)
    at r (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:1:73155)
    at m (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:1:74387)
    at http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:1:74459
    at Object.run (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:1:70959)
    at s.l.onStateChange (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:1:71435)
    at h (http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:6:43893)
    at http://127.0.0.1:8010/9af04d80e0c50a994cf4.min.js:18:321469

johanliebs, 2019-11-08:
Happened during project inspection.

ivankravets, 2019-11-09:
Could you share somewhere `project/.pio/build/%env/sizedata.json`?

johanliebs, 2019-11-10:
Shared in Drive: [https://drive.google.com/open?id=1TYSSzE2CAkboLpKAdV4baUY-vG0ZjHxt](url)

:)

elyulka, 2019-11-11:
Provided sizedata.json file works on v3.0.0 without errors on my machine. 
Browser version is what i would like to know. If it's old then it can be caused by the missing `Array.flat()` method.

ivankravets, 2019-11-11:
@johanliebs do you use sizedata.json from a project which fails?

johanliebs, 2019-11-12:
@ivankravets 
Yes I checked again, definitely the right project selected with the right sizedata.json. 
[screen shot of setup](https://drive.google.com/open?id=10jY8OWjQwO0s3m6uyZB8QHeEIVKBVoLI)
[screen shot of error](https://drive.google.com/open?id=1Dz9AhKEOAt-3K0yJ48f653k-BxXwn7XG)
[and the sizedata.json again](https://drive.google.com/open?id=1CVWgSdZb9HWbrYLn_FI48i4yqdJhZWVC)
Anything else I can check?

johanliebs, 2019-11-12:
@elyulka I think you just solved the issue. I regularly update platformio, but not atom which was one release behind. I noticed that the chrome update in atom went from 66.* to 69.* and now it works!
Sorry, should have checked that 1st. In the middle of a large project and my mind is elsewhere.

commit 5bc08c7b (2019-11-12): Replaced Array.flat with a polyfill to support IE Edge.

// Resolve #45
closed 2019-11-12 by elyulka