platformio/platformio-home #47: Could not import Arduino project

gcharles81, 2019-11-09:
PIO Core Call Error: "The current working directory C:\\Users\\Family Galea\\Documents\\PlatformIO\\Projects\\191109-105555-lolin32 will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\Family Galea\\Documents\\PlatformIO\\Projects\\191109-105555-lolin32\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing lolin32 (platform: espressif32; board: lolin32; framework: arduino)\n\nVerbose mode can be enabled via `-v, --verbose` option\nMkdir(\"C:\\Users\\Family Galea\\Documents\\PlatformIO\\Projects\\191109-105555-lolin32\\.pio\\build\\lolin32\")\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin32.html\nPLATFORM: Espressif 32 1.11.1 > WEMOS LOLIN32\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\nLDF Modes: Finder ~ chain, Compatibility ~ soft\nManifestParserError: Could not parse manifest -> Expecting property name: line 14 column 3 (char 349):\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 148:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"C:\\Users\\Family Galea\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"C:\\Users\\Family Galea\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\n    exec _file_ in call_stack[-1].globals\n  File \"C:\\Users\\Family Galea\\.platformio\\platforms\\espressif32\\builder\\main.py\", line 210:\n    target_elf = env.BuildProgram()\n  File \"C:\\Users\\Family Galea\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n    return self.method(*nargs, **kwargs)\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 139:\n    _build_project_deps(env)\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 48:\n    project_lib_builder = env.ConfigureProjectLibBuilder()\n  File \"C:\\Users\\Family Galea\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n    return self.method(*nargs, **kwargs)\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 1029:\n    lib_builders = env.GetLibBuilders()\n  File \"C:\\Users\\Family Galea\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n    return self.method(*nargs, **kwargs)\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 954:\n    lb = LibBuilderFactory.new(env, lib_dir)\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 54:\n    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 111:\n    self._manifest = manifest if manifest else self.load_manifest()\n  File \"C:\\Users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 662:\n    return ManifestParserFactory.new_from_file(manifest_path).as_dict()\n  File \"c:\\users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\package\\manifest\\parser.py\", line 67:\n    return ManifestParserFactory.new(get_file_contents(path), t, remote_url)\n  File \"c:\\users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\package\\manifest\\parser.py\", line 117:\n    return globals()[clsname](contents, remote_url, package_dir)\n  File \"c:\\users\\family~1\\platfo~1\\penv\\lib\\site-packages\\platformio\\package\\manifest\\parser.py\", line 127:\n    raise ManifestParserError(\"Could not parse manifest -> %s\" % e)\n [FAILED] Took 1.51 seconds"

ivankravets, 2019-11-09:
Could you share your Arduino.INO file to reproduce this issue?

the-red-o, 2019-11-10:
Was able to build and upload last night still, today the same issue:

_> Executing task in folder SKMesh - Fix: C:\Users\Theo\.platformio\penv\Scripts\platformio.exe run <

Processing node-gps (platform: espressif32; board: ttgo-t-beam; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 1.11.1 > TTGO T-Beam
HARDWARE: ESP32 160MHz, 1.25MB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
ManifestParserError: Could not parse manifest -> Expecting , delimiter: line 5 column 5 (char 79):
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:   
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Theo\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Theo\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\Theo\.platformio\platforms\espressif32\builder\main.py", line 210:
    target_elf = env.BuildProgram()
  File "C:\Users\Theo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Theo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1029:
    lib_builders = env.GetLibBuilders()
  File "C:\Users\Theo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 954:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 54:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 111:
    self._manifest = manifest if manifest else self.load_manifest()
  File "C:\Users\Theo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 662:
    return ManifestParserFactory.new_from_file(manifest_path).as_dict()
  File "c:\users\theo\.platformio\penv\lib\site-packages\platformio\package\manifest\parser.py", line 67:
    return ManifestParserFactory.new(get_file_contents(path), t, remote_url)
  File "c:\users\theo\.platformio\penv\lib\site-packages\platformio\package\manifest\parser.py", line 117:
    return globals()[clsname](contents, remote_url, package_dir)
  File "c:\users\theo\.platformio\penv\lib\site-packages\platformio\package\manifest\parser.py", line 127:
    raise ManifestParserError("Could not parse manifest -> %s" % e)
================================================================================ [FAILED] Took 1.24 seconds ================================================================================_ 

I updated the Espressif 32  platform to 1.11.1 to try and see if it will resolve the problem, but still the same error.