platformio/platformio-home #73: missing my mapped network drives in the New/Open project dialog box

SPWC, 2019-11-15:
Need to be able to access and work in these shared directories. The dialog that comes up when i click on new project doesnt show any of the mapped drives but shows my OS, D:/ and E:/ drive. I am on Windows 10 Enterprise 64-bit.

ivankravets, 2019-11-16:
Could you help us to debug this issue? See current source code https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L161

We use `wmic logicaldisk get name,VolumeName` and `fsutil fsinfo drives` to get drives list. Coud you try these commands on your machine?

SPWC, 2019-11-19:
Tried above and this is what was outputted:
```
PS C:\Windows\system32> wmic logicaldisk get name,VolumeName
Name  VolumeName
C:    OS
D:
E:

PS C:\Windows\system32> fsutil fsinfo drives

Drives: C:\ D:\ E:\
```
None of these are my network drives which are (\\192.168.xxx.xxx)(S:/, T:/, U:/, & V:/)

`wmic ` may have a command to show mapped disks, cannot find it but still looking.

SPWC, 2019-11-19:
Disconnected, then remapped my network drive seems to have fixed it. Still doesn't show up in `wmic logicaldisk get name,VolumeName` and `fsutil fsinfo drives` but I am not gonna complain about that here.

closed 2019-11-19 by SPWC