platformio/platformio-home #75: Could not list directory

chffelix, 2019-11-15:
'utf-8' codec can't encode character '\udcad' in position 10185: surrogates not allowed

ivankravets, 2019-11-25:
What is your OS? `python --version`?

chffelix, 2019-11-25:
Python 3.7.1. But if I was running an anaconda environment at the time it probably was Python 3.6.8. I don't really remember how it ended up in this message.. it was like the automatic crash handling and it asked to report the error, so I did and it ended up here in github.

chffelix, 2019-11-25:
Oh, and I'm using Slackware Linux, by the way.

ivankravets, 2019-11-26:
What is your locale?

chffelix, 2019-11-26:
My locale is pt-BR (Brazilian portuguese)

lachlanA, 2020-01-26:
I hit the same problem with a filename,  with junk charters in it,
I used "convmv"   (I'm running Debian, amd64) to find the file name, and then check the file,
and rename it.

**INSTALL** convmv (Debian)
   _sudo apt install convmv_

**RUN**
   _convmv -i -f ISO-8859-1 -t UTF-8 "Target_Director"_

the -i flag will run convmv in interactive mode. and not change any thing unless you allow it to.
_Note add -r if you wont to sub-directors._ 
This is the link I used to find the info
  _https://community.spiceworks.com/topic/271363-how-do-i-remove-invalid-utf-8-characters-in-filenames-using-bash_

Lachlan

closed 2020-09-17 by stale[bot]