platformio/platformio-home #80: 3.1 beta - Project Configuration - unneccessary edits to platformio.ini

pfeerick, 2019-11-16:
Behaviour Change Request

I think this may go in the 'too hard basket' ... but still have to ask ;)

The (fantastic, btw!) project configuration GUI does some... Makes some questionable changes to the `platformio.ini` configuration file. I shorten the header text on a lot of mine, and also comment my libdeps lines, but by ... 

1. opening the configurator
2. adding a description
3. hitting save 

... this removes the comments, and re-adds the full header to the file. Also, as noted by another user, it sometimes adds 

```
lib_archive = yes
test_build_project_src = no
```
on it's own, even though these options haven't been touched. 

So is it possible to *not* change lines if they haven't been modified... i.e. don't strip the comments, and perhaps not make changes to the header. 

[image]

ivankravets, 2019-11-16:
> lib_archive = yes
> test_build_project_src = no

Please open a separate issue for this, this is a bug. We will not save boolean field to the config if their value is == default.

ivankravets, 2019-11-16:
> this removes the comments, 

Yes, this is a known issue. We don't preserve comments because Python's ConfigParser does not support this :(
Should we show a warning somewhere?

> and re-adds the full header to the file

We do this by default when saving config file.

pfeerick, 2019-11-16:
Separate issue opened for those boolean parameters.

A warning ~would be good~ is basically mandatory, as I just realised it strips *all* comments, not just the inlines I was using for `lib_deps`. That could be rather... frustrating... for many users. 

It may be worth investigating other configuration parsers as [configobj](https://pypi.org/project/configobj/) as that apparently does preserve comments, although may have also have undesirable requirements for multi-line entries? Food for thought. 

I suspected the header behaviour wouldn't change, was just worth mentioning ;)

ivankravets, 2019-11-16:
> configobj 

I saw this project, it's fully abandoned => https://github.com/DiffSK/configobj/releases

pfeerick, 2019-11-16:
Hm... it's last release _was_ in 2016... but there are commits through 2018 and 2019... still doesn't seem to be that actively supported. :cry: 

In a word... frack. 

Warning it is!

TD-er, 2019-11-28:
Can you add a description or comment tag to the env section?
Maybe that can also be used later for other purposes in PIO to maybe show in a tooltip or whatever fancy way to give more info on the env section.

That would help overcome the lost comments.
A single line and multi line option would be great.
One for a more short description and one for something like an abstract.
It would also be useful to be able to wrap them in a define in the accompanying Python script so you can use it in the build itself.

pfeerick, 2019-11-29:
Oh... NICE! I like that thinking... maybe a `comments`/`notes` multiline and a single line `description` field that allows us to still document anything we want to document but not have to fight with ConfigParser?

ivankravets, 2019-11-30:
> Can you add a description or comment tag to the env section?

Yes, see https://github.com/platformio/platformio-home/issues/137

TD-er, 2019-11-30:
> > Can you add a description or comment tag to the env section?
> 
> Yes, see #137

That's not what I meant (or I don't understand your answer).
What I meant was something like this:

```
[env:MCVE_ESP8266_4M1M]
env_comment               = Simple MCVE environment
env_description           = Simple MCVE environment to build POC for new reported issues
platform                  = ${beta_platform.platform}
extends                   = esp8266_4M1M
```

ivankravets, 2019-11-30:
Ahhh, formatting? We do not do this. We use native Python's ConfigParser to save the configuration in a raw INI file. They know better how to do this.

If you need custom formatting, custom spaces, tabs, comments, etc., please use manual editing. This configuration is intended for people who do not want to play with INI files. Like me, and others. 

There is no difference for me in how Python will format it. It's the same when you do configuration in operating system UI. Is it important for you how OS saves XML, Plist and other files?

TD-er, 2019-11-30:
> Ahhh, formatting? We do not do this. We use native Python's ConfigParser to save the configuration in a raw INI file. They know better how to do this.

Nope, that's also not what I meant.
See these added (not existing) parameters:
```
env_comment = Simple MCVE environment
env_description = Simple MCVE environment to build POC for new reported issues
```
Just to make sure it is not about formatting, I removed the spaces ;)

ivankravets, 2019-11-30:
Sorry, now I understand. You can use custom fields in environment sections. Just add `custom_` prefix. For example, `custom_env_comment = Simple MCVE environment`.

andreoliwa, 2021-02-26:
FYI: try `ConfigUpdater` ([pyscaffold/configupdater: 🌴 Parser like ConfigParser but for updating configuration files](https://github.com/pyscaffold/configupdater)).

It works for me.
I stumbled into this issue when I was looking for a package to update `setup.cfg` and preserve comments on my package: [andreoliwa/nitpick: Enforce the same configuration across multiple projects](https://github.com/andreoliwa/nitpick).