platformio/platformio-home #86: Could not import project

ceuti, 2019-11-17:
PIO Core Call Error: "The current working directory C:\\Users\\Pedro\\Documents\\PlatformIO\\Projects\\191117-053108-counter will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\Pedro\\Documents\\PlatformIO\\Projects\\191117-053108-counter\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing icestick (platform: lattice_ice40; board: icestick)\n-------------------------------------------------------------------------------\nCorePackageManager: Installing tool-scons @ ~2.20501.7\ntool-scons @ 2.20501.190826 has been successfully installed!\nVerbose mode can be enabled via `-v, --verbose` option\n\n*** Import of non-existent variable ''projenv''\nFile \"C:\\Users\\Pedro\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 191, in <module>\n========================= [FAILED] Took 4.07 seconds ========================="

ceuti, 2019-11-17:
And when i try to upload a project to ICEZUM ALHAMBRA I, i get this message that it can foud .bilf file

ivankravets, 2019-11-17:
Do you use the latest PlatformIO Core 4.1? Please run "PlatformIO icon (Left Sidebar) > Misc > Upgrade PlatformIO Core". Restart Windows. Does it work now?

ceuti, 2019-11-17:
Yes i do. I get other message 
PIO Core v4.1.0 is used (previous was 4.1.1b1)\r\nPlease remove multiple PIO Cores from a system:\r\nhttps://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system\r\n
and i dont know that i have to do

ivankravets, 2019-11-25:
Could you provide a simple project to reproduce this issue?

ivankravets, 2019-12-02:
@guitarrista672f do you use the latest PlatformIO Core? I've just tried to create a new project and can't reproduce this issue.

ceuti, 2019-12-12:
Hi again. Sorry for delay answering.
When i try to open a simple project to ICE ZUM ALHAMBRA  i get this message
PIO Core Call Error: "The current working directory C:\\Users\\Pedro\\Documents\\PlatformIO\\Projects\\project1 will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\Pedro\\Documents\\PlatformIO\\Projects\\project1\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nPlatformManager: Installing lattice_ice40\nlattice_ice40 @ 1.2.2 has been successfully installed!\nPackageManager: Installing toolchain-icestorm @ ~1.10.0\r\ntoolchain-icestorm @ 1.10.1 has been successfully installed!\nPackageManager: Installing toolchain-iverilog @ ~1.1.0\r\ntoolchain-iverilog @ 1.1.0 has been successfully installed!\nThe platform 'lattice_ice40' has been successfully installed!\r\nThe rest of packages will be installed automatically depending on your build environment.\r\n\n\nError: Processing icezum (platform: lattice_ice40; board: icezum)\n-------------------------------------------------------------------------------\nCorePackageManager: Installing tool-scons @ ~2.20501.7\ntool-scons @ 2.20501.190826 has been successfully installed!\nVerbose mode can be enabled via `-v, --verbose` option\n---> WARNING: no .pcf file found\n\n*** Import of non-existent variable ''projenv''\nFile \"C:\\Users\\Pedro\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 191, in <module>\n========================= [FAILED] Took 12.46 seconds ========================="