platformio/platformio-home #122: Libraries: Search

Darlingtone, 2019-11-26:
**PIO Core Call Error: "Error: [API] Could not connect to PlatformIO API Service. Please try later."**

I keep getting the above error trying to install Libraries or Boards, the Board have since stopped but for Libraries am still getting it, this have so far prevented me from using PlatformIO as this has been ongoing for about a year now.

I have disabled my firewall and the only Antivirus on my system is Windows 10 Defender 

Any ideal how to resolve this issue?

ivankravets, 2019-11-26:
Please provide an output from system Terminal `cmd`
```
python --version
pip --version
pip list
```

Darlingtone, 2019-11-26:
Output from the cmd

```
C:\>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

C:\>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

```

ivankravets, 2019-11-26:
Just use a terminal in VSCode. PlatformIO icon (left sidebar) > Misc.. > New Terminal

Darlingtone, 2019-11-26:
```
pip 19.3.1 from c:\users\darlington\.platformio\penv\lib\site-packages\pip (python 3.7)

Python 3.7.4

Package          Version
---------------- ---------
bottle           0.12.17
certifi          2019.9.11
chardet          3.0.4
Click            7.0
colorama         0.4.1
idna             2.8
marshmallow      2.20.5
pip              19.3.1
platformio       4.1.0
pyelftools       0.25
pyserial         3.4
requests         2.22.0
semantic-version 2.8.3
setuptools       40.8.0
tabulate         0.8.6
urllib3          1.25.7
wheel            0.33.6
```

closed 2019-11-28 by Darlingtone