platformio/platformio-home #128: Save config causes silent error

elyulka, 2019-11-27:
Versions:
```
Home 3.1.0-beta.2
Core 4.1.1b3
```

Saving project config causes silent error:

[image]

@ivankravets Some issue on the Python side? I don't see any visible problems in the file:
https://raw.githubusercontent.com/TD-er/MCVE_ESPxx/master/platformio.ini

ivankravets, 2019-11-27:
@elyulka  Can you show errors to the user? So, they can report us directly.

ivankravets, 2019-11-27:
The saving issue has been fixed in https://github.com/platformio/platformio-core/commit/47469e87594d8067ce9728cc0e40cf6902e5ad0b

elyulka, 2019-11-28:
Fixed in f8472948f06e1eb4893183da1f434287932cf8b3

closed 2019-11-28 by ivankravets