platformio/platformio-home #130: Could not load PIO Account information

mrnakhkash, 2019-11-27:
PIO Core Call Error: "DLL load failed: The specified procedure could not be found."

ivankravets, 2019-11-28:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ThePirat91, 2019-12-09:
Hello, I have the same problem.
Below my infos

`C:\Python3.7\Scripts\;C:\Python3.7\;C:\Python27\;C:\Python27\Scripts;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Python3.5\;C:\Python3.5\Scripts\;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Android\platform-tools-latest-windows;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Overlook Fing 3.0\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\PuTTY\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\Acronis\VirtualFile\;C:\Program Files (x86)\Common Files\Acronis\VirtualFile64\;C:\Program Files (x86)\Common Files\Acronis\FileProtector\;C:\Program Files (x86)\Common Files\Acronis\FileProtector64\;C:\Program Files (x86)\Common Files\Acronis\SnapAPI\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PowerShell\7-preview\preview;C:\Program Files\PowerShell\6\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Git\cmd;C:\Ruby24\bin;C:\Ruby24-x64\bin;C:\Users\Cristian\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Nmap;`

`Python 3.7.2`

`pip 19.3.1 from c:\python3.7\lib\site-packages\pip (python 3.7)`

`altgraph              0.16.1
arcade                2.1.7
attrs                 19.3.0
bcrypt                3.1.7
bottle                0.12.18
certifi               2019.11.28
cffi                  1.13.2
chardet               3.0.4
Click                 7.0
colorama              0.4.1
colorlog              4.0.2
cycler                0.10.0
docutils              0.15.2
ecdsa                 0.14.1
esphome               1.14.3
esptool               2.8
future                0.18.2
idna                  2.8
ifaddr                0.1.6
Kivy                  1.11.1
kivy-deps.angle       0.1.9
kivy-deps.glew        0.1.12
kivy-deps.gstreamer   0.1.17
kivy-deps.sdl2        0.1.22
Kivy-examples         1.11.1
Kivy-Garden           0.1.4
kiwi-gtk              3.0.3
kiwisolver            1.1.0
marshmallow           2.20.5
matplotlib            3.1.2
MouseInfo             0.1.2
numpy                 1.17.4
opencv-contrib-python 4.1.2.30
paho-mqtt             1.5.0
passlib               1.7.2
pefile                2019.4.18
Pillow                6.2.1
pip                   19.3.1
platformio            4.1.0
protobuf              3.11.1
pyaes                 1.6.1
PyAutoGUI             0.9.48
pycparser             2.19
pyelftools            0.26
pygame                1.9.6
PyGetWindow           0.0.8
pyglet                1.4.8
pyglet-ffmpeg2        0.1.15
Pygments              2.5.2
PyInstaller           3.5
PyMsgBox              1.0.7
pyparsing             2.4.5
pyperclip             1.7.0
pypiwin32             223
Pypubsub              4.0.3
PyQt5                 5.13.2
PyQt5-sip             12.7.0
PyRect                0.1.4
PyScreeze             0.1.25
pyserial              3.4
PySide2               5.13.2
PySimpleGUI           4.8.0
PySimpleGUIQt         0.30.0
PySimpleGUIWeb        0.33.0
PySimpleGUIWx         0.14.0
python-dateutil       2.8.1
pytiled-parser        0.9.0
PyTweening            1.0.3
pytz                  2019.3
pywin32               227
pywin32-ctypes        0.2.0
PyYAML                5.2
remi                  2019.11
requests              2.22.0
semantic-version      2.8.3
setuptools            42.0.2
shiboken2             5.13.2
six                   1.13.0
tabulate              0.8.6
tornado               6.0.3
tzlocal               2.0.0
urllib3               1.25.7
voluptuous            0.11.7
wheel                 0.33.6
wxPython              4.0.7.post2
yolk3k                0.9`

Thanks!

closed 2020-09-17 by stale[bot]