platformio/platformio-home #133: Could not import project

zhangyu74, 2019-11-28:
PIO Core Call Error: "The current working directory /home/zhangyu/Documents/PlatformIO/Projects/191128-162212-arduino-blink will be used for the project.\n\nThe next files/directories have been created in /home/zhangyu/Documents/PlatformIO/Projects/191128-162212-arduino-blink\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError:"

ivankravets, 2019-11-28:
Could you reproduce this issue when restarting VSCode?

jatg81, 2020-04-28:
I have the same issue, it only happens when I tried to import esp-idf-arduino blink & esp-idf-arduino wifiscan.

valeros, 2020-04-28:
Hi @jatg81 ! Stange, but I cannot reproduce the problem. Could you please explain in more detail what were your steps when you encountered this issue?

jatg81, 2020-05-03:
Hi @valeros, Now the problem was solved. But I am not sure what was the solution exactly . I spent 2 days trying to solved it, I reinstalled many times vscode and PIO and also tried Atom with PIO and the problem appeared in both. Therefore I realized that it was not a vscode problem and it was something related with PIO. By chance I figured out that a plataform update for espress32 was available so I installed it and then I was able to import esp-idf-arduino blink & esp-idf-arduino.

closed 2020-09-17 by stale[bot]