platformio/platformio-home #137: Show one-time warning before saving a project configuration

ivankravets, 2019-11-28:
```
Warning! Project Configuration does not preserver comments and custom formatting in `platformio.ini` file. The file contents will be overridden with a current configuration which you see in UI. Would you like to save the configuration?

[Cancel] [Save] 
```

TD-er, 2019-11-29:
> The file contents will be overridden with a current configuration which you see in UI. Would you like to save the configuration?

This sentence can be improved.

> **Warning!** 
> The entire file contents or `platformio.ini` will be rewritten with the current configuration defined in this UI.
> Continue to save the configuration?

pfeerick, 2019-11-29:
Another approach to this might also be 'show the warning if the file has not been automatically generated' ... i.e. rather than be a one-time warning, show warning if the file has not been UI generated, and you're trying to make UI changes. If that is feasible. Consider making the warning hidable, for those that think they know what they're doing! ;)

ivankravets, 2019-11-30:
>  'show the warning if the file has not been automatically generated' 

If there is no `platformio.ini` file, you will not be able to edit it :) So, we show in Projects section only PlatformIO-based projects that have `platformio.ini`.

I'm not sure that it's easy to track how file was created before. I think this warning is good for every developer even he edits the first time. He will know that there is a difference between manual editing and UI.

closed 2019-12-02 by ivankravets

DevinNorgarb, 2022-06-12:
@ivankravets @TD-er 
> This sentence can be improved.

Do you not mean? 
> The entire file contents ~~or~~ **of** platformio.ini will be rewritten with the current configuration defined in this UI.

I came looking for this because I noticed it now in the IDE and thought that it should be corrected. 
Let me know if it should be `of` instead of `or`. and I quickly create a PR.

ivankravets, 2022-06-13:
Thanks! Solved in https://github.com/platformio/platformio-home/commit/531e4d226acf4d73d19444d365c90f4a5496f343

DevinNorgarb, 2022-06-13:
@ivankravets I was hoping to get my first commit into platform.io this way ;) 
But either way, glad I could help. 👍🏻

PR #151 (2019-12-02): Require users to consent to their own responsibility for side effects while  editing project config via UI
// Resolve #137
commit bc32e64f (2019-12-02): Implemented saga helper to display consent message if user has not yet (#151)

consented some terms.

// Resolve #137
commit 531e4d22 (2022-06-13): Typo fix // Resolve #137