platformio/platformio-home #141: Could not import project

yifanff, 2019-11-29:
PIO Core Call Error: "The current working directory C:\\Users\\liyifan\\Documents\\PlatformIO\\Projects\\191129-153401-VG will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\liyifan\\Documents\\PlatformIO\\Projects\\191129-153401-VG\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing imu330 (platform: aceinna_imu; board: OpenIMU330)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/aceinna_imu/OpenIMU330.html\r\nPLATFORM: Aceinna IMU 1.2.1 > Aceinna OpenIMU 330\r\nHARDWARE: STM32L431CB 80MHz, 64KB RAM, 128KB Flash\r\nDEBUG: Current (stlink) External (blackmagic, jlink, stlink)\r\nPACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)\r\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\r\nLDF Modes: Finder ~ chain, Compatibility ~ soft\r\nLooking for openIMU330-base-library library in registry\r\nFound: https://platformio.org/lib/show/6666/OpenIMU330-base-library\r\nLibraryManager: Installing id=6666 @ 1.0.4\r\nStopIteration: :\r\n  File \"C:\\Users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 151:\r\n    env.SConscript(\"$BUILD_SCRIPT\")\r\n  File \"C:\\Users\\liyifan\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 605:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\liyifan\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 286:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\liyifan\\.platformio\\platforms\\aceinna_imu\\builder\\main.py\", line 125:\r\n    target_elf = env.BuildProgram()\r\n  File \"C:\\Users\\liyifan\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 140:\r\n    _build_project_deps(env)\r\n  File \"C:\\Users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 48:\r\n    project_lib_builder = env.ConfigureProjectLibBuilder()\r\n  File \"C:\\Users\\liyifan\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 1027:\r\n    project.install_dependencies()\r\n  File \"C:\\Users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 859:\r\n    lm.install(uri)\r\n  File \"c:\\users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 301:\r\n    force=force,\r\n  File \"c:\\users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 698:\r\n    pkg_dir = self._install_from_piorepo(name, requirements)\r\n  File \"c:\\users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 151:\r\n    requirements,\r\n  File \"c:\\users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 478:\r\n    dlpath = self.download(url, tmp_dir, sha1)\r\n  File \"c:\\users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 182:\r\n    fd.start(with_progress=with_progress)\r\n  File \"c:\\users\\liyifan\\.platformio\\penv\\lib\\site-packages\\platformio\\downloader.py\", line 91:\r\n    f.write(next(itercontent))\r\n========================= [FAILED] Took 41.13 seconds ========================="

ivankravets, 2019-11-29:
Could you reproduce this issue after VSCode restart?

yifanff, 2019-12-12:
yes, i could reproduce this issue sometimes.

SwingW, 2020-04-16:
How did you solve the above problem，@yifanff

closed 2020-09-17 by stale[bot]