platformio/platformio-home #144: Platforms: Registry

victorfernandesraton, 2019-12-01:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 175, in platform_search\r\n    platform[\"name\"], with_boards=False, expose_packages=False\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 130, in _get_registry_platform_data\r\n    for p in _get_registry_platforms():\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 57, in _get_registry_platforms\r\n    platforms = util.get_api_result(\"/platforms\", cache_valid=\"7d\")\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 342, in get_api_result\r\n    result = cc.get(cache_key)\r\n  File \"C:\\Users\\vitol\\.platformio\\penv\\lib\\site-packages\\platformio\\app.py\", line 218, in get\r\n    with codecs.open(cache_path, \"rb\", encoding=\"utf8\") as fp:\r\n  File \"C:\\Users\\vitol\\AppData\\Local\\Programs\\Python\\Python37-32\\lib\\codecs.py\", line 898, in open\r\n    file = builtins.open(filename, mode, buffering)\r\nFileNotFoundError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\vitol\\\\.platformio\\\\.cache\\\\4f\\\\6a5bb1552c731dc7f8a58da257840c4f'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

victorfernandesraton, 2019-12-01:
I'm try install and using my arduino Uno and see this error.

ivankravets, 2019-12-02:
Please try to remove `C:\\Users\\vitol\\.platformio` folder and restart VSCode. Looks like antivirus tools block PlatformIO.

closed 2019-12-02 by ivankravets