platformio/platformio-home #148: Could not initialize project

guitarrista672f, 2019-12-01:
PIO Core Call Error: "The current working directory C:\\Users\\Fernando\\Documents\\PlatformIO\\Projects\\led33 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Fernando\\Documents\\PlatformIO\\Projects\\led33\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing icestick (platform: lattice_ice40; board: icestick)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\n---> WARNING: no .pcf file found\r\n\r\n*** Import of non-existent variable ''projenv''\r\nFile \"C:\\Users\\Fernando\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 191, in <module>\r\n========================== [FAILED] Took 1.97 seconds =========================="

guitarrista672f, 2019-12-01:
I just started to install visual code and platformio in order to work with stick kit evaluation ice40 but each time that I try to made a project I have an error message. I noted that this happen also with the alhambra 2 board.  If I use a different board there are not issues. I installed on two different computers the visual code and platformio with the same error .
Thanks

ivankravets, 2019-12-02:
Duplicate of https://github.com/platformio/platformio-home/issues/86

closed 2019-12-02 by ivankravets