platformio/platformio-home #154: Could not load platform data

assa62, 2019-12-04:
PIO Core Call Error: "'utf-8' codec can't decode byte 0xcf in position 175: invalid continuation byte"

ivankravets, 2020-02-29:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

assa62, 2020-02-29:
echo %PATH%
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\PROGRAM FILES (X86)\
INTEL\ICLS CLIENT\;C:\PROGRAM FILES\INTEL\ICLS CLIENT\;C:\Windows\SYSTEM32;C:\Wi
ndows;C:\Windows\SYSTEM32\WBEM;C:\Windows\SYSTEM32\WINDOWSPOWERSHELL\V1.0\;;C:\P
ROGRAM FILES\INTEL\WIFI\BIN\;C:\PROGRAM FILES\COMMON FILES\INTEL\WIRELESSCOMMON\
;C:\PROGRAM FILES (X86)\COMMON FILES\LENOVO;C:\PROGRAM FILES (X86)\COMMON FILES\
ULEAD SYSTEMS\MPEG;C:\SWTOOLS\READYAPPS;C:\PROGRAM FILES (X86)\INTEL\INTEL(R) MA
NAGEMENT ENGINE COMPONENTS\DAL;C:\PROGRAM FILES\INTEL\INTEL(R) MANAGEMENT ENGINE
 COMPONENTS\DAL;C:\PROGRAM FILES (X86)\INTEL\INTEL(R) MANAGEMENT ENGINE COMPONEN
TS\IPT;C:\PROGRAM FILES\INTEL\INTEL(R) MANAGEMENT ENGINE COMPONENTS\IPT;C:\PROGR
AM FILES\OPENVPN\BIN;C:\PROGRAM FILES\THINKPAD\BLUETOOTH SOFTWARE\;C:\PROGRAM FI
LES\THINKPAD\BLUETOOTH SOFTWARE\SYSWOW64;C:\Program Files\dotnet\;C:\Users\Alex\
AppData\Local\Programs\Python\Python38-32\Scripts\;C:\Users\Alex\AppData\Local\P
rograms\Python\Python38-32\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Co
mmon Files\Intel\WirelessCommon\;C:\Users\Alex\AppData\Local\Programs\Microsoft
VS Code\bin

echo %PYTHONPATH%
%PYTHONPATH%

python --version
Python 3.8.0

pip --version
pip 19.2.3 from c:\users\alex\appdata\local\programs\python\python38-32\lib\site
-packages\pip (python 3.8)

pip list

ivankravets, 2020-02-29:
Thanks! Could you reproduce this issue again?

assa62, 2020-02-29:
Сказать честно, я вообще не понимаю, как это(мои вопросы) здесь оказалось :(

closed 2020-03-01 by ivankravets