platformio/platformio-home #161: Could not load PIO Account information

cselti, 2019-12-06:
PIO Core Call Error: "the JSON object must be str, not 'bytes'"

cselti, 2019-12-06:
Probably my fault - works fine through pio cli.

cselti, 2019-12-07:
I got this message in PlatformIO IDE PIO Account GUI, but with the CLI I can login succesfully

ThePirat91, 2019-12-09:
Hello, I have your same problem but I cannot login via CLI either.

`C:\Users\Cristian>pio account register
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 33, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 11, in <module>
  File "C:\Users\Cristian\.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in <module>
    from twisted.internet._win32serialport import SerialPort
  File "C:\Users\Cristian\.platformio\packages\contrib-pysite\twisted\internet\_win32serialport.py", line 19, in <module>
    import win32file, win32event
ImportError: DLL load failed: The specified procedure could not be found.`

ivankravets, 2020-02-29:
@ThePirat91 Please uninstall all Pythons from a system and install the latest following this guide https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

closed 2020-02-29 by ivankravets