platformio/platformio-home #167: Libraries: Could not install library

emilio74, 2019-12-09:
PIO Core Call Error: "Library Storage: /home/emilio/.platformio/lib\nLibraryManager: Installing id=1089 @ ^2.7.1\nDownloading...\n\n\nError: Please read http://bit.ly/package-manager-ioerror\nError: Traceback (most recent call last):\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/urllib3/response.py\", line 425, in _error_catcher\n    yield\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/urllib3/response.py\", line 507, in read\n    data = self._fp.read(amt) if not fp_closed else b\"\"\n  File \"/usr/lib/python3.7/http/client.py\", line 457, in read\n    n = self.readinto(b)\n  File \"/usr/lib/python3.7/http/client.py\", line 501, in readinto\n    n = self.fp.readinto(b)\n  File \"/usr/lib/python3.7/socket.py\", line 589, in readinto\n    return self._sock.recv_into(b)\nConnectionResetError: [Errno 104] Connection reset by peer\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/requests/models.py\", line 750, in generate\n    for chunk in self.raw.stream(chunk_size, decode_content=True):\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/urllib3/response.py\", line 564, in stream\n    data = self.read(amt=amt, decode_content=decode_content)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/urllib3/response.py\", line 529, in read\n    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)\n  File \"/usr/lib/python3.7/contextlib.py\", line 130, in __exit__\n    self.gen.throw(type, value, traceback)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/urllib3/response.py\", line 443, in _error_catcher\n    raise ProtocolError(\"Connection broken: %r\" % e, e)\nurllib3.exceptions.ProtocolError: (\"Connection broken: ConnectionResetError(104, 'Connection reset by peer')\", ConnectionResetError(104, 'Connection reset by peer'))\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/click/decorators.py\", line 17, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/commands/lib.py\", line 151, in lib_install\n    library, silent=silent, interactive=interactive, force=force\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/managers/lib.py\", line 301, in install\n    force=force,\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/managers/package.py\", line 698, in install\n    pkg_dir = self._install_from_piorepo(name, requirements)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/managers/lib.py\", line 151, in _install_from_piorepo\n    requirements,\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/managers/package.py\", line 478, in _install_from_url\n    dlpath = self.download(url, tmp_dir, sha1)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/managers/package.py\", line 197, in download\n    raise e\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/managers/package.py\", line 182, in download\n    fd.start(with_progress=with_progress)\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/platformio/downloader.py\", line 91, in start\n    f.write(next(itercontent))\n  File \"/home/emilio/.platformio/penv/lib/python3.7/site-packages/requests/models.py\", line 753, in generate\n    raise ChunkedEncodingError(e)\nrequests.exceptions.ChunkedEncodingError: (\"Connection broken: ConnectionResetError(104, 'Connection reset by peer')\", ConnectionResetError(104, 'Connection reset by peer'))\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2020-02-29:
1. Please check your Internet connection
2. Try installing the library again.

closed 2020-02-29 by ivankravets

Peter-haotian-cui, 2020-02-29:
Thanks, it worked