platformio/platformio-home #173: Could not initialize project when try to create a new project for Makerdiary nRF52840 MDK with Zephyr

geeks-r-us, 2019-12-11:
PIO Core Call Error: "The current working directory /home/steve/Documents/PlatformIO/Projects/USBKeyboard will be used for the project.\n\nThe next files/directories have been created in /home/steve/Documents/PlatformIO/Projects/USBKeyboard\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nPlatformManager: Installing nordicnrf52\nDownloading\nDownloading\nUnpacking\nUnpacking\nnordicnrf52 @ 3.8.0 has been successfully installed!\nPackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0\ntoolchain-gccarmnoneeabi @ 1.70201.0 is already installed\nPackageManager: Installing tool-sreccat @ ~1.164.0\nDownloading\nDownloading\nUnpacking\nUnpacking\ntool-sreccat @ 1.164.0 has been successfully installed!\nThe platform 'nordicnrf52' has been successfully installed!\nThe rest of packages will be installed automatically depending on your build environment.\n\n\nError: Unknown board ID 'nrf52840_mdk'"

merlinschumacher, 2019-12-20:
I have the same issue. I also get the message "Unknown board ID 'nrf52840_mdk'" after adding the following to the platformio.ini
```
[env:nrf52840_mdk]
platform = nordicnrf52
board = nrf52840_mdk
```

iicky, 2019-12-23:
Same issue here on PlatformIO version Home 3.0.1 and Core 4.1.0.

ivankravets, 2020-02-29:
Duplicate of https://github.com/platformio/platformio-home/issues/155

closed 2020-02-29 by ivankravets