platformio/platformio-home #197: Could not load boards list

hpwit, 2019-12-21:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/commands/boards.py\", line 31, in cli\n    return _print_boards_json(query, installed)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/commands/boards.py\", line 77, in _print_boards_json\n    for board in _get_boards(installed):\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/commands/boards.py\", line 72, in _get_boards\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 190, in get_all_boards\n    boards = self.get_installed_boards()\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/util.py\", line 59, in wrapper\n    self.cache[key] = (time.time(), func(*args, **kwargs))\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 178, in get_installed_boards\n    p = PlatformFactory.newPlatform(manifest[\"__pkg_dir\"])\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 258, in newPlatform\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 514, in __init__\n    self._manifest = fs.load_json(manifest_path)\n  File \"/Users/yvesbazin/.platformio/penv/lib/python2.7/site-packages/platformio/fs.py\", line 78, in load_json\n    with open(file_path, \"r\") as f:\nIOError: [Errno 2] No such file or directory: '/Users/yvesbazin/.platformio/platforms/arduino-esp32/platform.json'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2020-02-29:
You have broken dev-platform. Please remove `%HOME_DIR%/.platformio/platforms` and install dev-platforms again.

closed 2020-02-29 by ivankravets

Shumatic, 2023-09-29:
Had the same issue with PIO 3.3.1 and C/C++1.17.5, and was about to abandon it.
But my solution was:
- Delete all extensions from within PIO (PIO and C/C++)
- Close vscode
- Delete C:\Users\xxxxxxx\.platformio
- Delete C:\Users\xxxxxxx\.vscode\extensions (this did the trick)
- Open vscode and reinstall PIO

Hope it helps someone