platformio/platformio-home #301: Libraries: Data

iqorqua, 2020-01-14:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 409, in lib_show\n    lib = util.get_api_result(\"/lib/info/%d\" % lib_id, cache_valid=\"1d\")\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 352, in get_api_result\n    cc.set(cache_key, result, cache_valid)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\site-packages\\platformio\\app.py\", line 242, in set\n    fp.write(data)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\codecs.py\", line 706, in write\n    return self.writer.write(data)\n  File \"c:\\users\\igorqua\\.platformio\\penv\\lib\\codecs.py\", line 370, in write\n    self.stream.write(data)\nIOError: [Errno 28] No space left on device\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

steveheinsch, 2020-01-14:
> [Errno 28] No space left on device\n
How big is the partition you're trying to upload to, and what is the size of the data?

iqorqua, 2020-01-15:
Oh, sorry, everything is fine, thats was because of Windows update :)
Thank you for quick response!

closed 2020-09-17 by stale[bot]