platformio/platformio-home #303: Allow copy/paste from Libraries->Example Code using hotkey

steveheinsch, 2020-01-14:
It's frustrating when being in the library section and clicking through the example code provided by a library, and not being able to just copy it. You can select it, but copying has no effect. So you can't just copy some of the example code and quickly paste it into your own sketch. You have to open the actual example file to do that, which is cumbersome next to just being able to copy/paste it when it's just sitting right there on the screen in front of you.

Edit: I just noticed if you go to the main application menu at the top of the screen and go to "Edit" and manually select "copy", it does copy it if it's highlighted, but using the CMD+C shortcut/hotkey (mac) does not, even though the CMD+C shortcut is listed right next to the "copy" menu item.

a-x-, 2020-03-15:
It's highly hateful problem for me!

I cannot copy even via vscode app menu! Context menu disabled at all.

Cmd-tab and other vscode wide hotkeys are not working too!

😩😭

Archer36, 2020-03-20:
Same problem here, drives me crazy. Seems like people have been talking about this one since 2018.

om2kw, 2020-07-28:
really nobody taking care, same problem here

closed 2020-09-19 by stale[bot]

a-x-, 2020-10-16:
Re-open please. It's unbelievable awful!