platformio/platformio-home #313: Could not initialize project

sosiax, 2020-01-18:
PIO Core Call Error:
"The current working directory /home/fons/MEGAsync/IoT/BLE400_T3/BLE400_T3 will be used for the project.

The next files/directories have been created in /home/fons/MEGAsync/IoT/BLE400_T3/BLE400_T3
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Error: Processing waveshare_ble400 (platform: nordicnrf51; board: waveshare_ble400; framework: zephyr)
--------------------------------------------------------------------------------
PackageManager: Installing framework-zephyr-open-amp @ ~0.20100.190612
framework-zephyr-open-amp @ 0.20100.190612 has been successfully installed!
PackageManager: Installing framework-zephyr-hal-nordic @ ~0.20100.191028
framework-zephyr-hal-nordic @ 0.20100.191028 has been successfully installed!
PackageManager: Installing tool-dtc @ ~1.4.7
tool-dtc @ 1.4.7 has been successfully installed!
PackageManager: Installing framework-zephyr-civetweb @ ~0.20100.190807
framework-zephyr-civetweb @ 0.20100.190807 has been successfully installed!
PackageManager: Installing toolchain-gccarmnoneeabi @ ~1.80201.0
toolchain-gccarmnoneeabi @ 1.80201.181220 has been successfully installed!
PackageManager: Installing framework-zephyr-mcumgr @ ~0.20100.190528
framework-zephyr-mcumgr @ 0.20100.190528 has been successfully installed!
PackageManager: Installing framework-zephyr-nffs @ ~0.20100.190523
framework-zephyr-nffs @ 0.20100.190523 has been successfully installed!
PackageManager: Installing framework-zephyr-mbedtls @ ~0.20100.191006
framework-zephyr-mbedtls @ 0.20100.191006 has been successfully installed!
PackageManager: Installing tool-cmake @ ~3.15.0
tool-cmake @ 3.15.5 has been successfully installed!
PackageManager: Installing framework-zephyr-openthread @ ~0.20100.191024
framework-zephyr-openthread @ 0.20100.191024 has been successfully installed!
PackageManager: Installing framework-zephyr-libmetal @ ~0.20100.190530
framework-zephyr-libmetal @ 0.20100.190530 has been successfully installed!
PackageManager: Installing tool-gperf @ ^3.0.0
tool-gperf @ 3.0.4 has been successfully installed!
PackageManager: Installing framework-zephyr @ ~1.20100.0
framework-zephyr @ 1.20100.200106 has been successfully installed!
PackageManager: Installing framework-zephyr-fatfs @ ~0.20100.190522
framework-zephyr-fatfs @ 0.20100.190522 has been successfully installed!
PackageManager: Installing framework-zephyr-littlefs @ ~0.20100.190811
framework-zephyr-littlefs @ 0.20100.190811 has been successfully installed!
PackageManager: Installing framework-zephyr-lvgl @ ~0.20100.190812
framework-zephyr-lvgl @ 0.20100.190812 has been successfully installed!
PackageManager: Installing tool-ninja @ ^1.7.0
tool-ninja @ 1.7.1 has been successfully installed!
PackageManager: Installing framework-zephyr-segger @ ~0.20100.190421
framework-zephyr-segger @ 0.20100.190421 has been successfully installed!
PackageManager: Installing framework-zephyr-tinycbor @ ~0.20100.191016
framework-zephyr-tinycbor @ 0.20100.191016 has been successfully installed!
PackageManager: Installing framework-zephyr-mipi-sys-t @ ~0.20100.191024
framework-zephyr-mipi-sys-t @ 0.20100.191024 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf51/waveshare_ble400.html
PLATFORM: Nordic nRF51 5.0.1 > Waveshare BLE400
HARDWARE: NRF51822 32MHz, 32KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: framework-zephyr-open-amp 0.20100.190612 (2.1.0), framework-zephyr-hal-nordic 0.20100.191028 (2.1.0), tool-dtc 1.4.7, framework-zephyr-civetweb 0.20100.190807 (2.1.0), toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1), framework-zephyr-mcumgr 0.20100.190528 (2.1.0), framework-zephyr-nffs 0.20100.190523 (2.1.0), framework-zephyr-mbedtls 0.20100.191006 (2.1.0), tool-cmake 3.15.5, framework-zephyr-openthread 0.20100.191024 (2.1.0), framework-zephyr-libmetal 0.20100.190530 (2.1.0), tool-gperf 3.0.4, framework-zephyr 1.20100.200106 (2.1.0), framework-zephyr-fatfs 0.20100.190522 (2.1.0), framework-zephyr-littlefs 0.20100.190811 (2.1.0), framework-zephyr-lvgl 0.20100.190812 (2.1.0), tool-ninja 1.7.1, tool-sreccat 1.164.0 (1.64), framework-zephyr-segger 0.20100.190421 (2.1.0), framework-zephyr-tinycbor 0.20100.191016 (2.1.0), framework-zephyr-mipi-sys-t 0.20100.191024 (2.1.0)
Installing Zephyrs Python dependenciesnDEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 wont be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting pyyaml
  Downloading https://files.pythonhosted.org/packages/3d/d9/ea9816aea31beeadccd03f1f8b625ecf8f645bd66744484d162d84803ce5/PyYAML-5.3.tar.gz (268kB)
Collecting pykwalify
  Downloading https://files.pythonhosted.org/packages/36/9f/612de8ca540bd24d604f544248c4c46e9db76f6ea5eb75fb4244da6ebbf0/pykwalify-1.7.0-py2.py3-none-any.whl (40kB)
Collecting python-dateutil>=2.4.2
  Downloading https://files.pythonhosted.org/packages/d4/70/d60450c3dd48ef87586924207ae8907090de0b306af2bce5d134d78615cb/python_dateutil-2.8.1-py2.py3-none-any.whl (227kB)
Collecting docopt>=0.6.2
  Downloading https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz
Collecting six>=1.5
  Downloading https://files.pythonhosted.org/packages/65/eb/1f97cb97bfc2390a276969c6fae16075da282f5058082d4cb10c6c5c1dba/six-1.14.0-py2.py3-none-any.whl
Building wheels for collected packages: pyyaml, docopt
  Building wheel for pyyaml (setup.py): started
  Building wheel for pyyaml (setup.py): finished with status done
  Created wheel for pyyaml: filename=PyYAML-5.3-cp27-cp27mu-linux_x86_64.whl size=45255 sha256=9bdbd47f9fdd0f9a2091c2152a848657ed90104637fd2101fa06701136eb8b5c
  Stored in directory: /home/fons/.cache/pip/wheels/e4/76/4d/a95b8dd7b452b69e8ed4f68b69e1b55e12c9c9624dd962b191
  Building wheel for docopt (setup.py): started
  Building wheel for docopt (setup.py): finished with status done
  Created wheel for docopt: filename=docopt-0.6.2-py2.py3-none-any.whl size=13704 sha256=a396b6f9e93fab2d0c0a53d5fafc914e8d991e1056d26cf85ded6b8bff41aaee
  Stored in directory: /home/fons/.cache/pip/wheels/9b/04/dd/7daf4150b6d9b12949298737de9431a324d4b797ffd63f526e
Successfully built pyyaml docopt
Installing collected packages: pyyaml, six, python-dateutil, docopt, pykwalify
Successfully installed docopt-0.6.2 pykwalify-1.7.0 python-dateutil-2.8.1 pyyaml-5.3 six-1.14.0
Python 3.4 or greater required, but you have Python 2.7.17
========================= [FAILED] Took 64.00 seconds ========================="

sosiax, 2020-01-18:
This error was reported...
At the bottom line it says that it detects only Python 2.7.17, but Python 3.7.5 is also installed and used as default.

gocamiot, 2020-01-20:
Same issue. Need help urgently. In apps on windows 10 removed any Python 2.7x references, only Python 3 references left.
Uninstall Platformio, VSStudio, reinstall, c:users/xxx/.platformio/penv has python 2.7 files in.
It seems it is not possible to install Platformio with VStudio choosing own python version?
Please help.

matsujirushi, 2020-01-25:
I got same error.

gocamiot, 2020-01-27:
Solved.
Uninstall/Remove all Python references from Windows 10.
Including Anaconda(if you have it.)
Install Windows x86-64 executable installer from site: https://www.python.org/downloads/release/python-381/
This is the kicker, YOU HAVE TO CHECK The link that says Add to Path.
After Python install remove penv directory in .platformio
Re open VStudio and your Platformio core should install, using python3.
And load zephyr examples!

GeorgeGkinis, 2020-03-11:
I am encountering the same issue. 

Python3 installed.
alias python=python3

Combination is win10 -> ubuntu subsystem -> zephyr

I see that /home/q/.local/lib/python2.7 has been created.

There seems to be no  penv directory on my system.

Any ideas?

closed 2020-09-17 by stale[bot]