platformio/platformio-home #348: Could not import project

Smrtokvitek, 2020-01-30:
PIO Core Call Error: "The current working directory C:\\Users\\Robert\\Documents\\PlatformIO\\Projects\\200130-130218-zephyr-blink will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\Robert\\Documents\\PlatformIO\\Projects\\200130-130218-zephyr-blink\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing blackpill_f103c8 (platform: ststm32; framework: zephyr; board: blackpill_f103c8)\n\nPackageManager: Installing framework-zephyr-hal-stm32 @ ~0.20100.190619\nDownloading\nDownloading\nUnpacking\nUnpacking\nframework-zephyr-hal-stm32 @ 0.20100.190619 has been successfully installed!\nPackageManager: Installing framework-zephyr-hal-st @ ~0.20100.190924\nDownloading\nDownloading\nUnpacking\nUnpacking\nframework-zephyr-hal-st @ 0.20100.190924 has been successfully installed!\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/blackpill_f103c8.html\nPLATFORM: ST STM32 6.0.0 > BlackPill F103C8\nHARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash\nDEBUG: Current (stlink) External (blackmagic, jlink, stlink)\nPACKAGES: framework-zephyr-open-amp 0.20100.190612 (2.1.0), tool-dtc 1.4.7, framework-zephyr-civetweb 0.20100.190807 (2.1.0), toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-zephyr-mcumgr 0.20100.190528 (2.1.0), framework-zephyr-libmetal 0.20100.190530 (2.1.0), framework-zephyr-nffs 0.20100.190523 (2.1.0), framework-zephyr-mbedtls 0.20100.191006 (2.1.0), tool-cmake 3.15.5, framework-zephyr-openthread 0.20100.191024 (2.1.0), framework-zephyr-hal-stm32 0.20100.190619 (2.1.0), framework-zephyr 1.20100.200106 (2.1.0), tool-ninja 1.9.0, framework-zephyr-fatfs 0.20100.190522 (2.1.0), framework-zephyr-lvgl 0.20100.190812 (2.1.0), framework-zephyr-hal-st 0.20100.190924 (2.1.0), framework-zephyr-littlefs 0.20100.190811 (2.1.0), framework-zephyr-segger 0.20100.190421 (2.1.0), framework-zephyr-tinycbor 0.20100.191016 (2.1.0), framework-zephyr-mipi-sys-t 0.20100.191024 (2.1.0)\nPython 3.4 or greater required, but you have Python 2.7.13\n [FAILED] Took 18.72 seconds \n\nEnvironment       Status    Duration\n----------------  --------  ------------\nblackpill_f103c8  FAILED    00:00:18.724\nnucleo_f030r8     IGNORED\nnucleo_l073rz     IGNORED\n 1 failed, 0 succeeded in 00:00:18.724"

Smrtokvitek, 2020-02-07:
This workaround works. When you delete python binary from the folder PlatformIO reinstalls  the python 3 instead during the next startup.

https://community.platformio.org/t/python-version-zephyr-blink-example-gives-compilation-error-in-the-pio-terminal/10996

valeros, 2020-06-25:
the Zephyr framework only supports Python 3.

closed 2020-06-25 by valeros