platformio/platformio-home #359: Libraries: Builtin

AarneAarebye, 2020-02-01:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/platformio/commands/lib.py\", line 383, in lib_builtin\n    return click.echo(dump_json_to_unicode(items))\n  File \"/Users/arne/.platformio/penv/lib/python2.7/site-packages/platformio/compat.py\", line 73, in dump_json_to_unicode\n    obj, encoding=get_filesystem_encoding(), ensure_ascii=False, sort_keys=True\n  File \"/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py\", line 251, in dumps\n    sort_keys=sort_keys, **kw).encode(obj)\n  File \"/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/encoder.py\", line 210, in encode\n    return ''.join(chunks)\nUnicodeDecodeError: 'ascii' codec can't decode byte 0xc4 in position 19: ordinal not in range(128)\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

AarneAarebye, 2020-02-01:
performed: pip install -U platformio --user
same issue

ivankravets, 2020-02-05:
Duplicate of https://github.com/platformio/platformio-core/issues/3370

closed 2020-02-05 by ivankravets