platformio/platformio-home #374: Bug: Installing libraries to project

pfeerick, 2020-02-07:
Installing a library to a project via the PIO Home Libraries 'Install to...' button does not appear to be doing anything. No entry is added to `platformio.ini`, nor is the library installed to the `.pio/libdeps` folder. 

There have been two reports against release versions of PIO 4.1, and I have been able to reproduce this behaviour myself (on PIO Home 3.1.0-rc.2 / Core 4.1.1b9). 
https://community.platformio.org/t/installing-librarys-to-specific-project-does-nothing/11782

https://community.platformio.org/t/installed-library-not-found-when-included/11882

ivankravets, 2020-03-06:
Sorry for the issue :( We fixed it before and plan to release PIO Core 4.2.2 soon.

closed 2020-03-06 by ivankravets

mriksman, 2020-03-06:
Confirmed working with 4.2.2a3

FaFyle, 2020-04-13:
It looks like it still doesn't work for me. Tried it with Adafruit GFX Library. I get the library downloaded in the .pio/lideps folder, but they stil aren't poping up in the platformio.ini file... So they are not inculded during compilation.