platformio/platformio-home #384: Could not import project

wsd07, 2020-02-10:
PIO Core Call Error: "The current working directory C:\\Users\\wsd07\\Documents\\PlatformIO\\Projects\\200210-154108-arduino-blink will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\wsd07\\Documents\\PlatformIO\\Projects\\200210-154108-arduino-blink\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing bluepill_f103c8 (platform: ststm32; framework: arduino; board: bluepill_f103c8)\r\n--------------------------------------------------------------------------------\r\nUnicodeDecodeError: 'utf-8' codec can't decode byte 0xb1 in position 0: invalid start byte:\r\n  File \"C:\\Users\\wsd07\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 88:\r\n    env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Defaults.py\", line 88:\r\n    _default_env = SCons.Environment.Environment(*args, **kw)\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 992:\r\n    apply_tools(self, tools, toolpath)\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 107:\r\n    env.Tool(tool)\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 1799:\r\n    tool(self)\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Tool\\__init__.py\", line 303:\r\n    self.generate(env, *args, **kw)\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Tool\\gcc.py\", line 59:\r\n    version = detect_version(env, env['CC'])\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Tool\\gcc.py\", line 94:\r\n    while SCons.Util.to_str(pipe.stdout.readline()):\r\n  File \"C:\\Users\\wsd07\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Util.py\", line 1593:\r\n    return str (s, 'utf-8')\r\n========================== [FAILED] Took 1.84 seconds ==========================\r\n\r\nEnvironment                 Status    Duration\r\n--------------------------  --------  ------------\r\nbluepill_f103c8             FAILED    00:00:01.841\r\nbluepill_f103c8_128k        IGNORED\r\nmaple                       IGNORED\r\ngeneric STM32F103RB         IGNORED\r\nbluepill_f103c8_stm32_core  IGNORED\r\nnucleo_l152re               IGNORED\r\nnucleo_f401re               IGNORED\r\nnucleo_wb55rg_p             IGNORED\r\nnucleo_h743zi               IGNORED\r\nnucleo_g071rb               IGNORED\r\nrobotdyn_blackpill_f303cc   IGNORED\r\nsparky_v1                   IGNORED\r\nvake_v1                     IGNORED\r\nst3dp001_eval               IGNORED\r\ndisco_f407vg                IGNORED\r\n==================== 1 failed, 0 succeeded in 00:00:01.841 ===================="

wsd07, 2020-02-10:
dose anybody know how to solve this problem?

ivankravets, 2020-02-12:
Could you try to install Python 3? See https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Then remove `C:\Users\wsd07\.platformio\penv` folder and restart VSCode.

wsd07, 2020-02-13:
> Could you try to install Python 3? See https://docs.platformio.org/en/latest/faq.html#install-python-interpreter
> 
> Then remove `C:\Users\wsd07\.platformio\penv` folder and restart VSCode.

thanks! I reinstalled platformio, and the problem was fixed

closed 2020-02-13 by wsd07

ivankravets, 2020-02-13:
Did you install Python 3?