platformio/platformio-home #398: Internal Store Exception

fabrmuller, 2020-02-14:
TypeError: Object.fromEntries is not a function
    at mapPropsToFields (http://127.0.0.1:8010/main.455fb94049687037e1ed.min.js:1:311791)
    at t.getInitialState (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:49:65683)
    at new t (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:19:335598)
    at xi (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:51168)
    at $a (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:67956)
    at xc (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:105945)
    at Pl (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:90235)
    at El (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:90160)
    at Ml (http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:87500)
    at http://127.0.0.1:8010/vendor.c2b02c04cda4b5d41dcb.min.js:57:45830

fabrmuller, 2020-02-14:
Error when I click on "Configure" button on a project in the "PlatformIO Home"
[image]

ivankravets, 2020-02-29:
Duplicate of https://github.com/platformio/platformio-home/issues/181

closed 2020-02-29 by ivankravets