platformio/platformio-home #414: Libraries: Could not install library

robstermn, 2020-02-18:
PIO Core Call Error: "Library Storage: C:\\Users\\veitchb\\.platformio\\lib\r\nLibraryManager: Installing url\r\nDownloading...\r\n\n\nError: Can not unpack file 'C:\\Users\\veitchb\\.platformio\\lib\\_tmp_installing-lvglc43s-package\\Servo'"

sysdl132, 2020-02-19:
I have the same PROBLEM!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
I cannot download because '404 not found'?????
There also has another:PR_CONNECT_RESET_ERROR.
What problem?????????

sysdl132, 2020-02-19:
And it shows installing... for a long time.

[image]

sysdl132, 2020-02-20:
(Anyone can fix it?I need to compile a sketch but the libraries is required.............)

closed 2020-02-29 by ivankravets