platformio/platformio-home #420: Libraries: Could not update library

Wikibear, 2020-02-21:
PIO Core Call Error: "Library Storage: C:\\Users\\me\\Documents\\PlatformIO\\Projects\\espurna-master\\espurna-master\\code\\.pio\\libdeps\\blitzwolf-bwshpx\r\nUpdating ArduinoJson                     @ 5.13.4         [6.14.1]\r\n\n\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 253, in lib_update\r\n    lm.update(library, only_check=only_check)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 805, in update\r\n    self.uninstall(pkg_dir, after_update=True)\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 723, in uninstall\r\n    with LockFile(self.package_dir):\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\lockfile.py\", line 102, in __enter__\r\n    self.acquire()\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\lockfile.py\", line 86, in acquire\r\n    return self._lock()\r\n  File \"c:\\users\\me\\.platformio\\penv\\lib\\site-packages\\platformio\\lockfile.py\", line 61, in _lock\r\n    self._fp = open(self._lock_path, \"w\")\r\nFileNotFoundError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\me\\\\Documents\\\\PlatformIO\\\\Projects\\\\espurna-master\\\\espurna-master\\\\code\\\\.pio\\\\libdeps\\\\blitzwolf-bwshpx.lock'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

sysdl132, 2020-02-22:
I have the same problem.

[image]

[image]

sysdl132, 2020-02-22:
I am working on Windows 10,1903

sysdl132, 2020-02-22:
@Wikibear I cannot download library,but can update  library.You can try the cmd or powershell.
If you are using VSCode:
```
cd C:\Users\<user_name>\.platformio\penv\Scripts
platformio.exe update
```

[image]

Wikibear, 2020-02-22:
@sysdl132 problem stay lib can't be updated:

[image]

ivankravets, 2020-02-29:
Duplicate of https://github.com/platformio/platformio-home/issues/408

closed 2020-02-29 by ivankravets