platformio/platformio-home #428: Libraries: Could not install library TaskScheduler by Anatoli Arkhipenko

aslansari, 2020-02-23:
PIO Core Call Error: "'utf-8' codec can't decode byte 0xf0 in position 771: invalid continuation byte"

ivankravets, 2020-02-29:
What is your OS?

aslansari, 2020-02-29:
Windows 10, version 3.1.2 works fine btw.

ivankravets, 2020-02-29:
How to reproduce this? I use Win10 in VM:
[image]

Do you install library into the global storage to into the project?

aslansari, 2020-02-29:
I installed the library into the global storage. However,
now I just retried and 3.1.3 has successfully installed, i'm not sure what caused the issue. I had clicked the button in the notification to report the issue at that time.

ivankravets, 2020-02-29:
Please reopen this issue if you will reproduce it. Thanks!

closed 2020-02-29 by ivankravets