platformio/platformio-home #442: Could not load boards list

imok1948, 2020-02-28:
PIO Core Call Error: "the JSON object must be str, not 'bytes'"

ivankravets, 2020-02-29:
What is your OS? Please provide an output from a system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

imok1948, 2020-02-29:
**Output of all commands is given below**

algomast@Lenovo-310:~/Documents/PlatformIO/Projects/nodemcu$ echo $PATH
/home/algomast/.platformio/penv/bin:/home/algomast/.platformio/penv:/home/algomast/bin:/home/algomast/.local/bin:/home/algomast/.nvm/versions/node/v8.15.1/bin:/usr/local/ant/bin:/home/algomast/bin:/home/algomast/.local/bin:/usr/local/ant/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-11-oracle/bin:/usr/lib/jvm/java-11-oracle/db/bin:/usr/lib/jvm/oracle_jdk8/bin:/usr/lib/jvm/oracle_jdk8/db/bin:/usr/lib/jvm/oracle_jdk8/jre/bin:/usr/lib/jvm/java-11-oracle/bin:/usr/lib/jvm/java-11-oracle/db/bin:/usr/lib/jvm/oracle_jdk8/bin:/usr/lib/jvm/oracle_jdk8/db/bin:/usr/lib/jvm/oracle_jdk8/jre/bin

algomast@Lenovo-310:~/Documents/PlatformIO/Projects/nodemcu$ python --version
Python 3.5.2

algomast@Lenovo-310:~/Documents/PlatformIO/Projects/nodemcu$ pip --version
pip 20.0.2 from /home/algomast/.platformio/penv/lib/python3.5/site-packages/pip (python 3.5)

algomast@Lenovo-310:~/Documents/PlatformIO/Projects/nodemcu$ pip list
Package          Version   
---------------- ----------
bottle           0.12.18   
certifi          2019.11.28
chardet          3.0.4     
Click            7.0       
colorama         0.4.3     
idna             2.9       
marshmallow      3.5.0     
pip              20.0.2    
pkg-resources    0.0.0     
platformio       4.2.2a1   
pyelftools       0.26      
pyserial         3.4       
requests         2.23.0    
semantic-version 2.8.4     
setuptools       20.7.0    
tabulate         0.8.6     
urllib3          1.25.8    
wheel            0.34.2

ivankravets, 2020-02-29:
What is your OS?

ivankravets, 2020-02-29:
Fixed in https://github.com/platformio/platformio-core/issues/3396

closed 2020-02-29 by ivankravets

imok1948, 2020-03-01:
> What is your OS?

Its Ubuntu 16.04

algomast@Lenovo-310:~$ lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:printing-9.20160110ubuntu0.2-amd64:printing-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial