platformio/platformio-home #445: Could not load recent projects

GeorgWolter, 2020-03-01:
Server error: {"args":[1,"Operation not permitted"],"message":"[Errno 1] Operation not permitted","type":"OSError"}

ivankravets, 2020-03-01:
Do you use antivirus tools?

ivankravets, 2020-03-01:
What is your OS?

GeorgWolter, 2020-03-01:
Hi Ivan, it's macOS 10.15.3

ivankravets, 2020-03-01:
How did you resovle it?

GeorgWolter, 2020-03-01:
I hope that I am not mixing up topics / problems.
Have to admit that I am new on VSC & Platformio but I already like it very much!
(My background is prof. HW & SW QA/QM and Infrastructure analysis in global TOP10 projects.)
Currently there is sometimes a problem with #include so including libraries. (red underline)
Even that libraries were installed the automatic code-check gives errors.
Interesting is: Such kind of problem suddenly disappeared yesterday.

GeorgWolter, 2020-03-02:
Started an new (easy) project tonight and the issue came up again.
Library "TheThingsNode.h" was installed project-related so not global.
At first only "TheThingsNode.h" had the red waveline; now even Arduino.h. 
Very strange...
[image]

ivankravets, 2020-03-02:
It looks like you have an issue reported here https://github.com/microsoft/vscode-cpptools/issues/4976

https://community.platformio.org/t/include-errors-detected/5858/2?u=ivankravets

Does it work if you open 1 project per 1 workspace?

GeorgWolter, 2020-03-02:
I can't get that under control. Highly motivated...but only burning hours and hours...
Thank you for escalating this issue :-)

ivankravets, 2020-03-02:
Does it work now?

GeorgWolter, 2020-03-02:
No success after creating a new workspace with one project item. Same trouble with INCLUDE.
But then success after removing local TTN Library and reinstallation as global.
I have 4 projects under 1 workspace w/o trouble but I am always sure that the INCLUDE issue
will come up again when I add another project. Will track that.

ivankravets, 2020-03-03:
Could you provide a simple project to reproduce this issue?

GeorgWolter, 2020-03-03:
Well  the issue came up very straighforwardly.
Using latest MacBook Pro 16" / macOS Catalina 10.15.3
- Installed VSC, installed PlatformIO
- Created a project, added an initial new project, added library / libraries
- Copy / paste of easy code into the project's main.cpp
VSC's current workflow in the Ui/uX for handling workspace(s) and projects is at least from my perspective inscrutable and difficult to understand; not up to date. 
I can offer a call e.g. by Skype (georg-wolter).

GeorgWolter, 2020-03-04:
I think the issue is connected to having file(s) open or not open when closing / restarting VSC.

GeorgWolter, 2020-03-04:
Released: 0.27.0-insiders

ivankravets, 2020-03-04:
Does it work with Released: 0.27.0-insiders?

GeorgWolter, 2020-03-04:
I cannot reproduce the issue with 0.27.0-insiders.

closed 2020-03-04 by ivankravets

GeorgWolter, 2020-03-07:
There is another (non-related) problem in "Quick Access" / field "New Project".
Copy / Paste by mouse-click right is not possible / functional
Can you confirm this? I am testing on a MacBook Pro16 with latest macOS.