platformio/platformio-home #479: Could not load platform updates

DavidFW1960, 2020-03-14:
PIO Core Call Error: "Error: VCS: Could not receive an output from `['git', 'ls-remote']` command ({'returncode': 128, 'err': u\"remote: Repository not found.\\nfatal: repository 'https://github.com/Jason2866/Arduino.git/' not found\", 'out': u''})"

clodoaldosouza, 2020-03-26:
Same problem: 
"PIO Core Call Error: "Error: VCS: Could not receive an output from `['git', 'ls-remote']` command ({'out': '', 'err': \"remote: Repository not found.\\nfatal: repository 'https://github.com/Jason2866/Arduino.git/' not found\\n\", 'returncode': 128})""

timbru31, 2020-04-13:
The problem is caused by having the entry to non-existing repo (i.e., https://github.com/Jason2866/Arduino.git). You can solve it by removing the folder which contains the repo above in your `~/.platformio` folder.

closed 2020-09-10 by DavidFW1960