platformio/platformio-home #525: Could not import project

simon-rob, 2020-03-25:
PIO Core Call Error: "The current working directory /home/simon/Documents/PlatformIO/Projects/200325-151250-INS will be used for the project.\n\nThe next files/directories have been created in /home/simon/Documents/PlatformIO/Projects/200325-151250-INS\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Unknown board ID 'openIMU300'"

valeros, 2020-03-25:
Hi @simon-rob ! This issue [has been fixed](https://github.com/Aceinna/platform-aceinna_imu/commit/5997102d95a8a310b97b3b5a05047c71133c16e4) in the development branch, the first letter of the board name should be in uppercase `OpenIMU300`.

ivankravets, 2020-09-09:
Please update all dev-platforms and restart VSCode.

closed 2020-09-09 by ivankravets