platformio/platformio-home #526: Could not initialize project

GrassHopper1977, 2020-03-25:
I got an update today and now I can't compile or even initialise ESP-IDF projects. Literal none of my existing projects build at all they all return various errors. I've tried uninstalling and reinstalling PlatformIO and Visual Studio. I've even tried creating a new project and that doesn't work either! I'm truly baffled! This is a result of an attempt to initialise a new project:

```
PIO Core Call Error: "The current working directory C:\\Users\\asmew\\Documents\\GitHub\\PRO0001-CAN-Phantom\\S0001-0020\\CAN Receiver 2 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\asmew\\Documents\\GitHub\\PRO0001-CAN-Phantom\\S0001-0020\\CAN Receiver 2\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 1.12.0 > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40000.200303 (4.0.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200226 (8.2.0)\r\nError: Detected a whitespace character in project paths.\r\n========================== [FAILED] Took 4.89 seconds =========================="
```

I'm running Windows X
Visual Studio Code
```
Version: 1.43.2 (user setup)
Commit: 0ba0ca52957102ca3527cf479571617f0de6ed50
Date: 2020-03-24T07:38:38.248Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18363
```

PlatformIO
```
Core 4.3.1·Home 3.1.1
```
It's unfortunate timing: I'm right in the middle of fixing a bug and starring at a looming deadline. I've never written an issue before, hope I'm doing this right.
I'm fresh out of ideas right now. Any help greatly appreciated!

valeros, 2020-03-25:
Hi @GrassHopper1977 ! Starting with `ESP-IDF v4.0` whitespace characters in paths are not allowed and looks like you have one in your project name `CAN Receiver 2`. As a workaround you can roll back to the previous version of the platform and finish your project before porting it to the latest version of ESP-IDF. e.g.:
```
; Custom stable version
[env:your_environment]
platform = esrepssif32@1.11.2
framework = espidf
board = esp32doit-devkit-v1
```

GrassHopper1977, 2020-03-25:
It seems that I can create projects in other folders. I was just starting to wonder about white space.
I'll try that now.
Thank you for your help.

GrassHopper1977, 2020-03-25:
That seems to be doing the trick for now. Thank you very much.
Can I just ask: What's the scope of the white space issue? I have a project with no spaces in the name that wasn't building but it was in a sub folder of a larger (private) GIT repo that does have a space in it's name. Would you expect that to generate an error too? (Help isn't clear on the subject)
Does v4.0 also have a problem with having sub directories under src? I was getting can't find file errors.

valeros, 2020-03-25:
It's hard to guess, but `v4.0` switched to CMake as the main build system and we also use it as the source of build configuration for our build system. We generate a basic `CMakeLists.txt` for legacy projects based on ESP-IDF v3.x, but CMake configuration can be tricky sometimes so you may need to manually edit CMake files to compile your projects successfully. That white space issue is the limitation of this new CMake build system, so we actually cannot do much about it.

GrassHopper1977, 2020-03-25:
OK I can work with the old version until I've got a chance to spend some time with this.
Thanks for the help

closed 2020-03-25 by GrassHopper1977