platformio/platformio-home #532: Could not initialize project

MeatMangler, 2020-03-26:
PIO Core Call Error: "The current working directory C:\\Users\\mike\\Documents\\PlatformIO\\Projects\\TTGO Debug will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\mike\\Documents\\PlatformIO\\Projects\\TTGO Debug\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing ttgo-lora32-v2 (platform: espressif32; board: ttgo-lora32-v2; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nError: Unknown board ID 'ttgo-lora32-v2'\r\n========================== [FAILED] Took 0.84 seconds =========================="

valeros, 2020-03-27:
Hi @MeatMangler ! Are you sure you're using the latest version of `espressif32` platform?

MeatMangler, 2020-03-27:
I did not.  I had 1.11.2 but an **upgrade to 1.12.0 fixed this**.  Many thanks and great work on this extension working with the ESP-Prog debugger.  I hoping this will become my default development environment (vscodium ide).

closed 2020-03-27 by MeatMangler