platformio/platformio-home #538: Could not load platform data

jmsimpson68, 2020-03-27:
PIO Core Call Error: "Platform /Users/jmsimpson/.platformio/platforms/espressif32\n--------\nUpdating espressif32                     @ 1.11.2         [1.12.0]\nUninstalling espressif32 @ 1.11.2: \t[OK]\nPlatformManager: Installing espressif32 @ 1.12.0\nespressif32 @ 1.12.0 has been successfully installed!\nUpdating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]\nUpdating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]\nUpdating framework-arduinoespressif32    @ 3.10004.200129 [Up-to-date]\nUpdating tool-mkspiffs                   @ 2.230.0        [Up-to-date]\n\n\nError: Traceback (most recent call last):\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py\", line 410, in platform_update\n    pm.update(platform, only_packages=only_packages, only_check=only_check)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 154, in update\n    self.cleanup_packages(list(p.packages))\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 167, in cleanup_packages\n    p = PlatformFactory.newPlatform(manifest[\"__pkg_dir\"])\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 273, in newPlatform\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 535, in __init__\n    self._manifest = fs.load_json(manifest_path)\n  File \"/Users/jmsimpson/.platformio/penv/lib/python2.7/site-packages/platformio/fs.py\", line 53, in load_json\n    with open(file_path, \"r\") as f:\nIOError: [Errno 2] No such file or directory: '/Users/jmsimpson/.platformio/platforms/espressif8266/platform.json'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

valeros, 2020-03-27:
Hi @jmsimpson68 ! Looks like you have broken `espressif8266` platform installed.  Try to delete in manually from here `/Users/jmsimpson/.platformio/platforms/espressif8266`

jmsimpson68, 2020-03-28:
Thank you for the reply!!!  I hope you’re all staying safe and entitling you’re zombie apocalypse!!

Jeff Simpson
571-212-0927

closed 2020-09-17 by stale[bot]