platformio/platformio-home #551: Could not initialize project

douglassinji, 2020-03-30:
PIO Core Call Error: "The current working directory C:\\POS_EMBARCADOS\\UNISAL\\UNISAL - Sistemas Operacionais e RTOS de Sistemas Embarcados\\Exemplo02 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\POS_EMBARCADOS\\UNISAL\\UNISAL - Sistemas Operacionais e RTOS de Sistemas Embarcados\\Exemplo02\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 1.12.0 > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40000.200303 (4.0.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200226 (8.2.0)\r\nInstalling ESP-IDF's Python dependencies\r\n'c:\\users\\douglas' nÆo ‚ reconhecido como um comando interno\r\nou externo, um programa oper vel ou um arquivo em lotes.\r\n*** Error 1\r\nError: Detected a whitespace character in project paths.\r\n========================== [FAILED] Took 1.71 seconds =========================="

valeros, 2020-03-30:
You have white space characters in project path `UNISAL - Sistemas Operacionais e RTOS de Sistemas Embarcados`, starting with `ESP-IDF v4.0` they  are not allowed.  More info [here](https://docs.platformio.org/en/latest/frameworks/espidf.html#configuration).

closed 2020-03-30 by valeros

douglassinji, 2020-03-30:
I've already changed the project name for many differents name, unistalled the visual code and also the platformio, but it keep showing the same error

> 
PIO Core Call Error: "The current working directory C:\\UNISAL\\teste will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\UNISAL\\teste\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 1.12.0 > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40000.200303 (4.0.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200226 (8.2.0)\r\nInstalling ESP-IDF's Python dependencies\r\n'c:\\users\\douglas' nÆo ‚ reconhecido como um comando interno\r\nou externo, um programa oper vel ou um arquivo em lotes.\r\n*** Error 1\r\nError: Detected a whitespace character in project paths.\r\n========================== [FAILED] Took 1.62 seconds =========================="
>

valeros, 2020-03-31:
What is your user name? Looks like it has another white space.