platformio/platformio-home #558: Could not initialize project

jwatte, 2020-03-31:
PIO Core Call Error: "********************************************************************************\r\nObsolete PIO Core v4.2.1 is used (previous was 4.3.1)\r\nPlease remove multiple PIO Cores from a system:\r\nhttps://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system\r\n********************************************************************************\r\n\r\nThe current working directory C:\\Users\\jwatte\\Documents\\PlatformIO\\Projects\\Test Teensy Malum Board will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\jwatte\\Documents\\PlatformIO\\Projects\\Test Teensy Malum Board\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Traceback (most recent call last):\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    :param epilog: like the help string but it's printed at the end of the\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    \"\"\"A command collection is a multi command that merges multiple multi\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    if cmd.hidden:\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    if context_settings is None:\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n    def pass_obj(f):\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 143, in project_init\r\n    pg.generate()\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 125, in generate\r\n    tpl_vars = self._load_tplvars()\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 78, in _load_tplvars\r\n    tpl_vars.update(load_project_ide_data(self.project_dir, self.env_name) or {})\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\helpers.py\", line 140, in load_project_ide_data\r\n    from platformio.commands.run.command import cli as cmd_run\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\run\\command.py\", line 24, in <module>\r\n    from platformio.commands.device.command import device_monitor as cmd_device_monitor\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\device\\command.py\", line 23, in <module>\r\n    from platformio.commands.device import helpers as device_helpers\r\n  File \"C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\device\\helpers.py\", line 22, in <module>\r\n    from platformio.compat import get_object_members, load_python_module\r\nImportError: cannot import name 'get_object_members' from 'platformio.compat' (C:\\Users\\jwatte\\.platformio\\penv\\lib\\site-packages\\platformio\\compat.py)\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

jwatte, 2020-03-31:
Trying to create a new PlatformIO Teensy 3.6 project from scratch, in default location.
I have run the PlatformIO: Update All action.

valeros, 2020-04-03:
Looks like you have multiple `platformio-core` installed in your system. Please try run in the terminal window `pio upgrade`

closed 2020-09-17 by stale[bot]