platformio/platformio-home #576: Windows breakage when defaultencoding for non-unicode applications is ...

bunnyevans, 2020-04-07:
W10 with default encoding for non-unicode programs = Japanese 
(cp932)

Attempt to create a project with esp-idf

= boom.

[bang1.txt](https://github.com/platformio/platformio-home/files/4442880/bang1.txt)

This appears to be because python is picking up the system encoding, going "oh, I'm JAPANESE" and then falling flat on its face. So much for unicode...

As a non-python-programmer I am unable to fix this short of having a completely separate windows machine that is entirely in English.

valeros, 2020-04-07:
Hi @bunnyevans ! Thanks for the report. Looks like, there is a problem in internal ESP-IDF tools and will be resolved in the next release. Please take a look a this issue https://github.com/espressif/esp-idf/issues/4977

closed 2020-04-07 by valeros

bunnyevans, 2020-04-07:
Fascinating, but ideally it should be starting python with whatever magical flags it needs to ignore systemlocale (cp932 et cetera) and running it in unicode.

Possibly like: https://docs.python.org/3/using/windows.html
Section 3.7