platformio/platformio-home #578: Could not initialize PIO IDF project in VSCode

durun-durun, 2020-04-07:
Hello everybody,
I would like to create a project with PIO in VS Code with IDF library. But  the following error message appeares. Could you please help me?
Thank you!

PIO Core Call Error: "The current working directory C:\\Users\\schwarzt\\Documents\\PlatformIO\\Projects\\project_project will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\schwarzt\\Documents\\PlatformIO\\Projects\\project_project\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\n-------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\nPLATFORM: Espressif 32 1.12.0 > Espressif ESP32 Dev Module\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40000.200303 (4.0.0) \n - tool-cmake 3.16.5 \n - tool-esptoolpy 1.20600.0 (2.6.0) \n - tool-idf 1.0.1 \n - tool-mconf 1.4060000.20190628 (406.0.0) \n - tool-ninja 1.9.0 \n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \n - toolchain-xtensa32 2.80200.200226 (8.2.0)\nReading CMake configuration...\n-- Could NOT find Git (missing: GIT_EXECUTABLE) \n-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32\n-- IDF_TARGET not set, using default target: esp32\n-- The C compiler identification is GNU 8.2.0\n-- The CXX compiler identification is GNU 8.2.0\n-- The ASM compiler identification is GNU\n-- Found assembler: C:/Users/schwarzt/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\n-- Check for working C compiler: C:/Users/schwarzt/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\n-- Check for working C compiler: C:/Users/schwarzt/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works\n-- Detecting C compiler ABI info\n-- Detecting C compiler ABI info - done\n-- Detecting C compile features\n-- Detecting C compile features - done\n-- Check for working CXX compiler: C:/Users/schwarzt/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe\n-- Check for working CXX compiler: C:/Users/schwarzt/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works\n-- Detecting CXX compiler ABI info\n-- Detecting CXX compiler ABI info - done\n-- Detecting CXX compile features\n-- Detecting CXX compile features - done\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\n-- Project version: 1\n-- Building ESP-IDF components for target esp32\n-- Checking Python dependencies...\n-- Configuring incomplete, errors occurred!\nSee also \"C:/Users/schwarzt/Documents/PlatformIO/Projects/project_project/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log\".\nCMake Warning at C:/Users/schwarzt/.platformio/packages/framework-espidf/tools/cmake/git_submodules.cmake:4 (message):\n  Git executable was not found.  Git submodule checks will not be executed.\n  If you have any build issues at all, start by adding git executable to the\n  PATH and rerun cmake to not see this warning again.\nCall Stack (most recent call first):\n  C:/Users/schwarzt/.platformio/packages/framework-espidf/tools/cmake/idf.cmake:34 (include)\n  C:/Users/schwarzt/.platformio/packages/framework-espidf/tools/cmake/project.cmake:7 (include)\n  CMakeLists.txt:2 (include)\n\n\nCMake Error at C:/Users/schwarzt/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):\n  Some Python dependencies must be installed.  Check above message for\n  details.\nCall Stack (most recent call first):\n  C:/Users/schwarzt/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)\n  C:/Users/schwarzt/.platformio/packages/framework-espidf/tools/cmake/project.cmake:337 (idf_build_process)\n  CMakeLists.txt:3 (project)\n========================= [FAILED] Took 30.57 seconds ========================="

closed 2020-09-17 by stale[bot]

mtraven, 2024-03-15:
did you ever solve this?  I'm trying to do the same thing & getting a similar error.  would love to know how you resolved it!

mtraven, 2024-03-15:
I know its 4 years later, but I just ran into this problem myself.  Turned out to be too long of file names(path).  I moved my projects folder to \esp\proj\ and that fixed it.  Hope that helps anyone else who stumbled upon this post!