platformio/platformio-home #580: Could not import Arduino project

d-popov, 2020-04-08:
PIO Core Call Error: "The current working directory C:\\Users\\popov\\Documents\\PlatformIO\\Projects\\200408-105759-esp01_1m will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\popov\\Documents\\PlatformIO\\Projects\\200408-105759-esp01_1m\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Traceback (most recent call last):\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 134, in project_init\r\n    init_base_project(project_dir)\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 174, in init_base_project\r\n    config.save()\r\n  File \"C:\\Users\\popov\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\config.py\", line 463, in save\r\n    with open(path or self.path, \"w+\") as fp:\r\nPermissionError: [Errno 13] Permission denied: 'C:\\\\Users\\\\popov\\\\Documents\\\\PlatformIO\\\\Projects\\\\200408-105759-esp01_1m\\\\platformio.ini'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

valeros, 2020-04-08:
```
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\popov\\Documents\\PlatformIO\\Projects\\200408-105759-esp01_1m\\platformio.ini
```
Please try another location for your project.

d-popov, 2020-04-08:
It was the BitDefender antivirus

closed 2020-04-08 by d-popov