platformio/platformio-home #582: Could not initialize project

tandrejko, 2020-04-09:
PIO Core Call Error: "The current working directory /Users/tomaszandrejko/Documents/PlatformIO/Projects/test will be used for the project.\n\nThe next files/directories have been created in /Users/tomaszandrejko/Documents/PlatformIO/Projects/test\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing uno (platform: atmelavr; board: uno; framework: arduino)\n--------------------------------------------------------------------------------\nPermissionError: [Errno 13] Permission denied: '/etc/paths.d/Wireshark':\n  File \"/Users/tomaszandrejko/.platformio/penv/lib/python3.7/site-packages/platformio/builder/main.py\", line 89:\n    env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)\n  File \"/Users/tomaszandrejko/.platformio/packages/tool-scons/script/../engine/SCons/Defaults.py\", line 88:\n    _default_env = SCons.Environment.Environment(*args, **kw)\n  File \"/Users/tomaszandrejko/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 951:\n    platform(self)\n  File \"/Users/tomaszandrejko/.platformio/packages/tool-scons/script/../engine/SCons/Platform/__init__.py\", line 129:\n    return self.generate(*args, **kw)\n  File \"/Users/tomaszandrejko/.platformio/packages/tool-scons/script/../engine/SCons/Platform/darwin.py\", line 59:\n    with open(file, 'r') as f:\n========================== [FAILED] Took 0.70 seconds =========================="

tandrejko, 2020-04-09:
What should I do ??

valeros, 2020-04-13:
Looks like `/etc/paths.d/Wireshark` file that is not readable by your user?

closed 2020-09-17 by stale[bot]