platformio/platformio-home #588: Could not initialize project

AlexanderBurtasov, 2020-04-10:
PIO Core Call Error: "The current working directory C:\\Users\\idle1\\OneDrive\\Документы\\PlatformIO\\Projects\\drone will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\idle1\\OneDrive\\Документы\\PlatformIO\\Projects\\drone\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 1.12.0 > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40000.200303 (4.0.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200226 (8.2.0)\r\nReading CMake configuration...\r\n-- Found Git: C:/Program Files/Git/cmd/git.exe (found version \"2.21.0.windows.1\") \r\n-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32\r\n-- IDF_TARGET not set, using default target: esp32\r\n-- The C compiler identification is GNU 8.2.0\r\n-- The CXX compiler identification is GNU 8.2.0\r\n-- The ASM compiler identification is GNU\r\n-- Found assembler: C:/Users/idle1/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/idle1/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/idle1/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works\r\n-- Detecting C compiler ABI info\r\n-- Detecting C compiler ABI info - done\r\n-- Detecting C compile features\r\n-- Detecting C compile features - done\r\n-- Check for working CXX compiler: C:/Users/idle1/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe\r\n-- Check for working CXX compiler: C:/Users/idle1/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works\r\n-- Detecting CXX compiler ABI info\r\n-- Detecting CXX compiler ABI info - done\r\n-- Detecting CXX compile features\r\n-- Detecting CXX compile features - done\r\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\r\n-- Project version: 1\r\n-- Building ESP-IDF components for target esp32\r\n-- Checking Python dependencies...\r\n-- Configuring incomplete, errors occurred!\r\nSee also \"C:/Users/idle1/OneDrive/Р”РѕРєСѓРјРµРЅС‚С‹/PlatformIO/Projects/drone/.pio/build/esp32doit-devkit-v1/CMakeFiles/CMakeOutput.log\".\r\n\r\nfatal: not a git repository (or any of the parent directories): .git\r\nCMake Error at C:/Users/idle1/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):\r\n  Some Python dependencies must be installed.  Check above message for\r\n  details.\r\nCall Stack (most recent call first):\r\n  C:/Users/idle1/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)\r\n  C:/Users/idle1/.platformio/packages/framework-espidf/tools/cmake/project.cmake:337 (idf_build_process)\r\n  CMakeLists.txt:3 (project)\r\n\r\n\r\n\r\n========================== [FAILED] Took 9.03 seconds =========================="

valeros, 2020-04-10:
Looks like you have non-ascii characters in path `Документы`. Could you please try to initialize a project in another directory?

FernandoTorresFilho, 2020-09-11:
i have the same problem, any solution?

(idf_build_process)\r\n CMakeLists.txt:3 (project)\r\n\r\n\r\n\r\n========================= [FAILED] Took..

Could not initialize project
PIO Core Call Error: "The current working directory C:\\ESP32\\new2 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\ESP32\\new2\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 (1.12.4) > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40001.200521 (4.0.1) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200827 (8.2.0)\r\nReading CMake configuration...\r\n-- Found Git: C:/Program Files/Git/cmd/git.exe (found version \"2.28.0.windows.1\") \r\n-- IDF_TARGET not set, using default target: esp32\r\n-- The C compiler identification is GNU 8.2.0\r\n-- The CXX compiler identification is GNU 8.2.0\r\n-- The ASM compiler identification is GNU\r\n-- Found assembler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works\r\n-- Detecting C compiler ABI info\r\n-- Detecting C compiler ABI info - done\r\n-- Detecting C compile features\r\n-- Detecting C compile features - done\r\n-- Check for working CXX compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe\r\n-- Check for working CXX compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works\r\n-- Detecting CXX compiler ABI info\r\n-- Detecting CXX compiler ABI info - done\r\n-- Detecting CXX compile features\r\n-- Detecting CXX compile features - done\r\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\r\n-- Project version: 1\r\n-- Building ESP-IDF components for target esp32\r\n-- Checking Python dependencies...\r\nPython requirements from C:\\Users\\ferna\\.platformio\\packages\\framework-espidf\\requirements.txt are satisfied.\r\n\r\n-- Found PythonInterp: C:/Users/ferna/.platformio/penv/Scripts/python.exe (found version \"3.7.7\") \r\n-- Could NOT find Perl (missing: PERL_EXECUTABLE) \r\n-- Adding linker script C:/ESP32/new2/.pio/build/esp32doit-devkit-v1/esp-idf/esp32/esp32_out.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp32/ld/esp32.project.ld.in\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp32/ld/esp32.peripherals.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld\r\n-- Components: __pio_env app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs src tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa\r\n-- Component paths: C:/ESP32/new2/.pio/build/esp32doit-devkit-v1/__pio_env C:/Users/ferna/.platformio/packages/framework-espidf/components/app_trace C:/Users/ferna/.platformio/packages/framework-espidf/components/app_update C:/Users/ferna/.platformio/packages/framework-espidf/components/asio C:/Users/ferna/.platformio/packages/framework-espidf/components/bootloader C:/Users/ferna/.platformio/packages/framework-espidf/components/bootloader_support C:/Users/ferna/.platformio/packag [... cut ...] latformio/packages/framework-espidf/tools/cmake/build.cmake:306 (include)\r\n C:/Users/ferna/.platformio/packages/framework-espidf/tools/cmake/build.cmake:451 (__build_process_project_includes)\r\n C:/Users/ferna/.platformio/packages/framework-espidf/tools/cmake/project.cmake:396 (idf_build_process)\r\n CMakeLists.txt:3 (project)\r\n\r\n\r\n\r\n========================= [FAILED] Took 16.60 seconds ========================="

Already have tried to define the location to C:/Esp32, but didn't work.

valeros, 2020-09-11:
@FernandoTorresFilho Looks like you're using an outdated version of `espressif32` platform. Please run `pio update` in the IDE terminal.

closed 2020-09-11 by valeros

FernandoTorresFilho, 2020-09-11:
> @FernandoTorresFilho Looks like you're using an outdated version of `espressif32` platform. Please run `pio update` in the IDE terminal.

Thank you very much for the fast reply.
You were correct, the version installed was 1.12.4, and i updated to 2.0.0,

but unfortunately the error persist..

PIO Core Call Error: "The current working directory C:\\ESP32\\testr will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\ESP32\\testr\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\r\nPLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40100.200827 (4.1.0) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200827 (8.2.0)\r\nReading CMake configuration...\r\n-- Found Git: C:/Program Files/Git/cmd/git.exe (found version \"2.28.0.windows.1\") \r\n-- The C compiler identification is GNU 8.2.0\r\n-- The CXX compiler identification is GNU 8.2.0\r\n-- The ASM compiler identification is GNU\r\n-- Found assembler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works\r\n-- Detecting C compiler ABI info\r\n-- Detecting C compiler ABI info - done\r\n-- Detecting C compile features\r\n-- Detecting C compile features - done\r\n-- Check for working CXX compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe\r\n-- Check for working CXX compiler: C:/Users/ferna/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works\r\n-- Detecting CXX compiler ABI info\r\n-- Detecting CXX compiler ABI info - done\r\n-- Detecting CXX compile features\r\n-- Detecting CXX compile features - done\r\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\r\n-- Project version: 1\r\n-- Building ESP-IDF components for target esp32\r\n-- Checking Python dependencies...\r\nPython requirements from C:\\Users\\ferna\\.platformio\\packages\\framework-espidf\\requirements.txt are satisfied.\r\n\r\n-- Found PythonInterp: C:/Users/ferna/.platformio/penv/Scripts/python.exe (found version \"3.7.7\") \r\n-- Could NOT find Perl (missing: PERL_EXECUTABLE) \r\n-- Adding linker script C:/ESP32/testr/.pio/build/esp32doit-devkit-v1/esp-idf/esp32/esp32_out.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp32/ld/esp32.project.ld.in\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp32/ld/esp32.peripherals.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld\r\n-- Adding linker script C:/Users/ferna/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld\r\n-- Components: __pio_env app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs src tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa\r\n-- Component paths: C:/ESP32/testr/.pio/build/esp32doit-devkit-v1/__pio_env C:/Users/ferna/.platformio/packages/framework-espidf/components/app_trace C:/Users/ferna/.platformio/packages/framework-espidf/components/app_update C:/Users/ferna/.platformio/packages/framework-espidf/components/asio C:/Users/ferna/.platformio/packages/framework-espidf/components/bootloader C:/Use [... cut ...] ckages/framework-espidf/tools/cmake/build.cmake:68 (git_describe)\r\n C:/Users/ferna/.platformio/packages/framework-espidf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)\r\n C:/Users/ferna/.platformio/packages/framework-espidf/tools/cmake/idf.cmake:48 (__build_init)\r\n C:/Users/ferna/.platformio/packages/framework-espidf/tools/cmake/project.cmake:7 (include)\r\n CMakeLists.txt:2 (include)\r\n\r\n\r\n\r\n========================= [FAILED] Took 16.72 seconds ========================="

valeros, 2020-09-11:
Really strange, are you importing a project or creating a new one?