platformio/platformio-home #589: Could not initialize project

alrabe, 2020-04-10:
To reproduce: Start VS Code. Install Platform IO Extension. Restart Code. Try to add an new "Arduino Uno" Project. Boom!

> PIO Core Call Error: "The current working directory C:\\Users\\alexa\\Documents\\PlatformIO\\Projects\\PWM_Fan will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\alexa\\Documents\\PlatformIO\\Projects\\PWM_Fan\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing uno (platform: atmelavr; board: uno; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\n  File \"C:\\Users\\alexa\\.platformio\\platforms\\atmelavr\\builder\\main.py\", line 30\r\n\r\n    print \"Please unplug/plug device ...\"\r\n\r\n                                        ^\r\n\r\nSyntaxError: Missing parentheses in call to 'print'. Did you mean print(\"Please unplug/plug device ...\")?\r\n\r\n========================== [FAILED] Took 0.53 seconds =========================="

valeros, 2020-04-10:
Looks like you're using an outdated version of the `atmelavr` platform.  Could you please try to delete `C:\Users\alexa\.platformio\platforms\atmelavr` folder and try again?

alrabe, 2020-04-10:
Ok - I could create the project now. But why the version was outdated. I'm not 100% sure but 90% I would say I installed Platform IO the first time.

closed 2020-04-10 by alrabe