platformio/platformio-home #606: Could not initialize project

Peter-haotian-cui, 2020-04-15:
PIO Core Call Error: "The current working directory /Users/peter/Documents/PlatformIO/Projects/STM32F405 will be used for the project.\n\nThe next files/directories have been created in /Users/peter/Documents/PlatformIO/Projects/STM32F405\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Unknown board ID 'adafruit_feather_f405'"

valeros, 2020-04-15:
Hi @Peter-haotian-cui ! What version of `ststm32` platform are you using?

Peter-haotian-cui, 2020-04-16:
Hi~ @valeros  I don't understand what you mean "version of ststm32". 
My platformio.ini is

```ini 
[env:adafruit_feather_f405]
platform = ststm32
board = adafruit_feather_f405
framework = arduino
upload_protocol = stlink
```

valeros, 2020-04-17:
Open `PlatformIO Home`, go to `Platforms` -> `Installed` -> `ST STM32`. Is there an update available?

Peter-haotian-cui, 2020-04-19:
Thanks! After I update the board, the Error gone.

closed 2020-04-19 by Peter-haotian-cui